riscv-non-isa/riscv-rvv-intrinsic-doc #19: Add functions to copy data between these vector register types and standard C arrays.

Hsiangkai, 2020-05-08:
Can you please add a few simple convenience functions for copying data between these vector register types and standard C arrays, e.g., to make debug and things like printf() of a vector easier?  I’m thinking something like:   int64_t foo[MVL]; foo = toArray(vs1);

rdolbeau, 2020-05-08:
... isn't that just the load/store intrinsics? (see the `printiv64`/`printfv64` in https://github.com/rdolbeau/EPI-test-codes-vector/blob/master/simplefft16-rvv/test_fft16_zgemv.c)

The 'easy' way for printf would be a vector 'Length modifier/Conversion specifiers' to signal vectors so that printf can deal with them directly. Would be convenient, but then users will request all the already existing modifiers (width, precision, display style, ...) and then request some brand new ones (field separator between vector lanes, ...) ...

Hsiangkai, 2020-05-19:
I agree with you. Vector store is enough.

closed 2020-06-01 by Hsiangkai