riscv-non-isa/riscv-rvv-intrinsic-doc #26: Examples need updating

rofirrim, 2020-06-19:
Hi,

we have been testing an early implementation of the intrinsics RFC against the two examples in this repository using a small emulator of ours. We have been able to run `rvv_saxpy.c` successfully using `LMUL=1` instead of LMUL=8 but we found it needed a couple of changes other than those changes to go from `m8` to `m1`.

```diff
diff --git a/rvv_saxpy.c b/rvv_saxpy.c
index 6b43025..a4de0f0 100644
--- a/rvv_saxpy.c
+++ b/rvv_saxpy.c
@@ -44,7 +44,7 @@ float output[N] = {
     0.2484350696132857};
 
 void saxpy_golden(size_t n, const float a, const float *x, float *y) {
-  for (size_t i; i < n; ++i) {
+  for (size_t i = 0; i < n; ++i) {
     y[i] = a * x[i] + y[i];
   }
 }
@@ -55,11 +55,11 @@ void saxpy_vec(size_t n, const float a, const float *x, float *y) {
   vfloat32m8_t vx, vy;
 
   for (; (l = vsetvl_e32m8(n)) > 0; n -= l) {
-    vx = vle_v_f32m8(x);
+    vx = vle32_v_f32m8(x);
     x += l;
-    vy = vle_v_f32m8(y);
+    vy = vle32_v_f32m8(y);
     vy = vfmacc_vf_f32m8(vy, a, vx);
-    vse_v_f32m8 (y, vy);
+    vse32_v_f32m8 (y, vy);
     y += l;
   }
 }
```

Also we understand some of the loads in `sgemm` needs updating to use intrinsics, like shown below:

```diff
diff --git a/rvv_sgemm.c b/rvv_sgemm.c
index 975ba99..71ebd28 100644
--- a/rvv_sgemm.c
+++ b/rvv_sgemm.c
@@ -49,7 +49,7 @@ float b_array[MAX_BLOCKSIZE] = {1.7491401329284098,  0.1325982188803279,
 float golden_array[OUTPUT_LEN];
 float c_array[OUTPUT_LEN];
 
-void sgemm_golden() {
+void sgemm_golden(void) {
   for (size_t i = 0; i < MLEN; ++i)
     for (size_t j = 0; j < NLEN; ++j)
       for (size_t k = 0; k < KLEN; ++k)
@@ -63,23 +63,24 @@ void sgemm_vec(size_t size_m, size_t size_n, size_t size_k,
                size_t ldb,
                float *c, // m * n matrix
                size_t ldc) {
-  int i, j, k;
+  int j, k;
   size_t vl;
   vfloat32m1_t vec_c;
   for (int i = 0; i < size_m; ++i) {
     j = size_n;
     const float *bnp = b;
     float *cnp = c;
-    for (; vl = vsetvl_e32m1(j); j -= vl) {
+    for (; (vl = vsetvl_e32m1(j)); j -= vl) {
       const float *akp = a;
       const float *bkp = bnp;
-      vec_c = *(vfloat32m1_t *)cnp;
+      vec_c = vle32_v_f32m1(cnp);
       for (k = 0; k < size_k; ++k) {
-        vec_c = vfmacc_vf_f32m1(vec_c, *akp, *(vfloat32m1_t *)bkp);
+        vec_c = vfmacc_vf_f32m1(vec_c, *akp,
+            vle32_v_f32m1(bkp));
         bkp += ldb;
         akp++;
       }
-      *(vfloat32m1_t *)cnp = vec_c;
+      vse32_v_f32m1(cnp, vec_c);
       cnp += vl;
       bnp += vl;
     }
@@ -98,7 +99,7 @@ int fp_eq(float reference, float actual, float relErr)
 int main() {
   // golden
   memcpy(golden_array, b_array, OUTPUT_LEN * sizeof(float));
-  sgemm_golden(MLEN, NLEN, KLEN, a_array, KLEN, b_array, NLEN, golden_array, NLEN);
+  sgemm_golden();
   // vector
   memcpy(c_array, b_array, OUTPUT_LEN * sizeof(float));
   sgemm_vec(MLEN, NLEN, KLEN, a_array, KLEN, b_array, NLEN, c_array, NLEN);
```

However I'm confused with the sgemm vector as it seems to load a row of the "b" matrix rather than a column? Should this load below

```cpp
            vle32_v_f32m1(bkp));
```

load a column instead using a strided load or I totally misunderstood the code?

Kind regards,

zakk0610, 2020-06-22:
Hi Roger,
Sorry , I forget to update examples to rvv 0.9 include removing vector c operator,
thanks for point out,  I will update it soon.

sgemm example is translated from rvv spec [example](https://github.com/riscv/riscv-v-spec/blob/master/example/sgemm.S) without loop unrolling, you can reference below diagram to understand the code.
[image]

Thanks.

rofirrim, 2020-06-22:
Hi @zakk0610,

thanks for the prompt answer. Definitely I was mistaken with the sgemm.

Kind regards,

rofirrim, 2020-06-22:
Hi @zakk0610,

some extra comments in the hope they're useful.

```diff
-void sgemm_golden() {
+void sgemm_golden(void) {
   for (size_t i = 0; i < MLEN; ++i)
     for (size_t j = 0; j < NLEN; ++j)
       for (size_t k = 0; k < KLEN; ++k)
-        c_array[i * NLEN + j] += a_array[i * KLEN + k] * b_array[j + k * NLEN];
+        golden_array[i * NLEN + j] += a_array[i * KLEN + k] * b_array[j + k * NLEN];
 }
```
```diff
 int main() {
   // golden
-  memcpy(golden_array, b_array, OUTPUT_LEN * sizeof(float));
-  sgemm_golden(MLEN, NLEN, KLEN, a_array, KLEN, b_array, NLEN, golden_array, NLEN);
+  sgemm_golden();
   // vector
-  memcpy(c_array, b_array, OUTPUT_LEN * sizeof(float));
   sgemm_vec(MLEN, NLEN, KLEN, a_array, KLEN, b_array, NLEN, c_array, NLEN);
```

With these changes (along with those of the load/store) this sgemm test passes in our emulator.

Thanks!

zakk0610, 2020-06-23:
Thanks @rofirrim! 
I fixed them and they can also passed in spike simulator now.

closed 2020-06-23 by zakk0610

kito-cheng, 2020-06-23:
https://github.com/riscv/rvv-intrinsic-doc/commit/fa898dfe5e5284f533ba2c3e3c103ace24bf05dd