riscv-non-isa/riscv-rvv-intrinsic-doc #28: How to combine/split vectors use rvv intrinsics?

joy2myself, 2020-07-01:
Hi,

I want to use rvv intrinsics to combine/split vectors.
Combining operation is like the vcombine instruction in NEON, to contact two short vectors.
And splitting operation is something like vget_low and vget_high instructions in NEON, to get halves of the vector. 
But I didn't find intrinsics to combine/split vectors in the reference manual.
I think maybe I can use shift operation and some arithmetic operations to realize these two functions, but it is too complicated. Is there an easy way to do this? Or should we add such a function to intrinsics?

Sincerely,
Yin

joy2myself, 2020-07-05:
I asked this question at riscv-v-spec and as Krste mentioned, I can combine two vectors as bellow:

```
vsetvli  t0, 8, e8,m1
vle8.v  v1 src1
vle8.v  v2 src2
vsetvli  t0, 16, e8,m1
vslideup.vi  v1, v2, 64
```

But now I want to use rvv intrinsics in C++ to implement this function.
And the intrinsic of `vslideup` instruction is defined as bellow:

`vint8m1_t vslideup_vx_i8m1 (vint8m1_t src, size_t offset);`

I can not pass the dest vector (v1) to `vslideup`, so they are not combined. How can I use intrinsics to implement this assembly code? Should the dest vector parameter be added in the `vslideup` intrinsic?

Sincerely,
Yin

zakk0610, 2020-07-06:
Hi Yin,
We can use mask vslideup with all-ones mask, like
```
vslideup_vx_i8m1_m (vmset_m_b8() , v1, vint8m1_t v2, 64);
```
Ideally compiler can do optimization to generate `vslideup.vi` without mask.
Thanks.

joy2myself, 2020-08-12:
Hi,
There may be some problems with my previous description. In fact, what I want to do is
for example to combine two `vint8mf2_t` to one `vint8m1_t` and to split one `vint8m1_t` to two `vint8mf2_t`. It looks just like is a register allocation problem and do not need vector instructions. Are there intrinsics working for this?

ebahapo, 2020-08-12:
Should the intrinsic for `vslide{up,down}` take the destination as an argument, since its previous contents are partially preserved by the operation?  This is easy to express in assembly, but the intrinsics abstract this.

zakk0610, 2020-08-15:
@joy2myself  Why not to combine two `vint8m1_t` to one `vint8m1_t` like Krste's example?

joy2myself, 2020-08-16:
@zakk0610 Hi Zakk. In fact, the input I met is two `vint8mf2_t` but not `vint8m1_t`. If I use `vslide{up,down}` to combine, I should firstly append zero to these two `vint8mf2_t` to convert them into `vint8m1_t`. This can also be regarded as a kind of combination (combine `vint8mf2_t` and a full zero `vint8mf2_t`).
And also, split is hard to implement with `vslide{up,down}`, right?

In my opinion, this kind of combination is just a register alloction problem. We just need to put them together and rewrite the `vtype` register. We do not need to use vector instructions.

I think combination and split are very useful functions to make conversions between multiple short vectors and one long vector. For example two `__mf2` and one `__m1` or two `__m1` and one `__m2`. And these functions are independent of vector assembly instructions. Should we add these intrinsics? Is there any other way to implement it？

kito-cheng, 2020-08-17:
For fractional LMUL, it's not just register allocation issue, because fractional LMUL still occupy one register, if you want to combine two vector values with fractional LMUL, you need to slide up one of the value and then combine with another one.

For LMUL >= 1 case, it's just register allocation issue ***only if*** vl == vlmax, consider vlmax = 4 and vl = 3 (for LMUL=1), v0 (m1) = 1, 2, 3, 4 and v1 (m1) = 5, 6, 7, 8, combine two register you will get v0 (m2) = 1, 2, 3, 4, 5, 6, 7, 8, not 1, 2, 3, 5, 6,7.

joy2myself, 2020-08-17:
Hi Kito, thanks a lot for the guidance.
Then how could I implement combination and split? For example from two `vint8mf2_t` to one `vint8m1_t`? Do I need to store them first and then use C/C++ for combination? `vslide{up,down}` works with the same LMUL vector type. And I don't find intrinsics to convert `vint8mf2_t` to `vint8m1_t`.

Hsiangkai, 2020-09-07:
There are discussions converting to different LMUL under the same SEW in #37.

eopXD, 2022-07-29:
We have `lmul_trunc`, `lmul_ext`, `vget` and `vset` in the intrinsics to match your use.

closed 2022-07-29 by eopXD