riscv-non-isa/riscv-rvv-intrinsic-doc #30: Is The Mask Register Layout in rvv intrinsics different with v spec 0.9?

joy2myself, 2020-07-16:
Hi, I noticed that the Mask Register Layout has changed in riscv-v-spec version 0.9.

> A vector mask occupies only one vector register regardless of SEW and LMUL. Each element is allocated a single mask bit in a mask vector register.
> The mask bit for element _i_ is located in bit _i_ of the mask register, independent of SEW or LMUL.

But if I don't get it wrong, the design of mask types in rvv intrinsics still use a suffix _n_ (_n_=SEW/LMUL) to tell the MLEN.

Should this part be modified to be consistent with the v0.9 spec? Is there any design content in progress that can be shared?

sincerely,
Yin Zhang

zakk0610, 2020-07-17:
Hi Yin,
In HW view the layout is different between 0.9 and 0.8,
but in software view, the same ratio `(SEW/LMUL)` means the same number of bit used in mask register.
For example if `VLEN=128`, config with `SEW=32, LMUL=4` and `SEW=64, LMUL=8` uses the same number of 16 bit (`128*4/32` and `128*8/64`) in a mask vector register.

Thanks.

joy2myself, 2020-07-17:
Thank you for your explanation Zakk. I think I understand now. The ratio _n_=`SEW/LMUL` is used to show the number of bits mask used in register but not the MLEN.

closed 2020-07-21 by Hsiangkai