riscv-non-isa/riscv-rvv-intrinsic-doc #32: Add a new argument to ensure the result is correct for vslideup and vslidedown

monkchiang, 2020-07-24:
All of vslideup and vslidedown have an issue.  
For example, the vslideup.vx operation is vd[i + rs1] = vs2[i].
So We don't know Vd's value from vd[i] to vd[i + rs1]. 
If we want to know all of vd's value, then we need to insert a argumenet,
to initial Vd value. Thus,  I suggest to add a new argument for initial Vd value.

`vint8m1_t vslideup_vx_i8m1 (vint8m1_t src, size_t offset);`

change to 

`vint8m1_t vslideup_vx_i8m1 (vint8m1 dst, vint8m1_t src, size_t offset);`

kito-cheng, 2020-07-24:
This issue is similar to https://github.com/riscv/rvv-intrinsic-doc/issues/25 , so I think it's reasonable changes, @rofirrim any comment?

rofirrim, 2020-07-24:
I agree. I think it makes sense to be able able to reliably know what is going to be in the unchanged elements.

commit 5a84ca1b (2020-08-05): [RFC] Fix #32. Provide the original value to vector slide intrinsics.
closed 2020-08-05 by Hsiangkai