riscv-non-isa/riscv-rvv-intrinsic-doc #33: How to define vzero and vreinterpret

WilliamWangPeng, 2020-07-26:
Dear all,    
&ensp; &ensp;&ensp; We are adding new intrinsics which is depending on riscv-v-spec 0.9 recently,  but there remain two new intrinsics "vzero and vreinterpret" which I have never seen in riscv-v-spec 0.9.      
&ensp; &ensp;&ensp; The definition of vzero and  vreinterpret we have defined is as following in 
 src/llvm/include/llvm/IR/IntrinsicsRISCVVector.td
```  
def int_riscv_vzero_i8m1
  : Intrinsic<[llvm_nxv1i8_ty],
              [],
              [IntrNoMem]>;  

def int_riscv_vreinterpret_u64_i64_u64m2
  : Intrinsic<[llvm_nxv2i64_ty],
              [llvm_nxv2i64_ty],
              [IntrNoMem]>;
```
&ensp; &ensp;&ensp; But in src/llvm/include/llvm/IR/RISCVInstrInfoV.td,  the riscv-v-spec 0.9 doesn't give any illustration of the encoding definition to vzero and  vreinterpret, Do you know how to define them?    
```  
//__builtin_riscv_vzero_i8m1()
def : Pat<(int_riscv_vzero_i8m1), ()>;

//___builtin_riscv_vreinterpret_u64_i64_u64m2(src)
def : Pat<(int_riscv_vreinterpret_u64_i64_u64m2), ()>;
```    
Thank you  
Best Regards  
William

rdolbeau, 2020-07-26:
Those intrinsics don't correspond to an actual instruction. They are here to help with writing high-level code.

`̀vzero` outputs an all-zero vector (of the appropriate type), i.e. __builtin_riscv_vzero_i8m1() outputs a zero-valued `vint8m1_t`. The compiler has some freedom on how to achieve that (using a zero-generating idiom like `xor a,a,a`, reusing a previously-known-to-be-zero vector, etc.)

`vreinterpret` like ___builtin_riscv_vreinterpret_i64_u64m2 merely changes the high-level types without changing any bit in the vector. In this case it merely changes the 'signedness' of the type, reinterpreting `vint64m2_t` to `vuint64m2_t`. 

Both are described as 'utilisy functions' in https://github.com/riscv/rvv-intrinsic-doc/blob/master/rvv-intrinsic-rfc.md#utility-functions

WilliamWangPeng, 2020-07-26:
> Those intrinsics don't correspond to an actual instruction. They are here to help with writing high-level code.
>
> `̀vzero` outputs an all-zero vector (of the appropriate type), i.e. [...]

Thank you for your reply, dear Romain Dolbeau
I have known the use of vzero and vreinterpret, so due to all of these definition, we don't need to define the vzero and vreinterpret pat in src/llvm/include/llvm/IR/RISCVInstrInfoV.td like int_riscv_vle8_v_i8m1, right?
```  
//__builtin_riscv_vzero_i8m1()
def : Pat<(int_riscv_vzero_i8m1), ()>;

//___builtin_riscv_vreinterpret_u64_i64_u64m2(src)
def : Pat<(int_riscv_vreinterpret_u64_i64_u64m2), ()>;  
```

closed 2020-08-01 by WilliamWangPeng