riscv-non-isa/riscv-rvv-intrinsic-doc #35: Why is there no intrinsics for Type-Convert from 'vfloat32m1_t' to 'vfloat16mf2_t' in Narrowing Floating-Point Type-Convert intrinsics?

joy2myself, 2020-08-19:
I find that there is no Narrowing Floating-Point Type-Convert intrinsics for from 'vfloat32m1_t' to 'vfloat16mf2_t' in [Intrinsic Functions List](https://github.com/riscv/rvv-intrinsic-doc/blob/master/rvv_intrinsic_funcs.md)

Like this:
`vfloat16mf2_t vfncvt_f_f_w_f16mf2 (vfloat32m1_t src);`

Is it missing? Or is there no such intrinsic for some reason?

kito-cheng, 2020-08-19:
We have :
- 'vfloat32mf2_t' to 'vfloat16mf4_t' 
- 'vfloat32m2_t' to 'vfloat16m1_t'
- 'vfloat32m4_t' to 'vfloat16m2_t'
- 'vfloat32m8_t' to 'vfloat16m4_t'

but there is no 'vfloat32m1_t' to 'vfloat16mf2_t'  and 'vfloat32mf4_t' to 'vfloat16mf8_t'  so I believe it's mistake :P Thanks your report!

joy2myself, 2020-08-19:
Thanks for reply.

closed 2020-08-19 by joy2myself