riscv-non-isa/riscv-rvv-intrinsic-doc #36: Support vxrm intrinsics

HanKuanChen, 2020-08-25:
May we have intrinsics to set `vxrm`?
Follow [fenv](https://en.cppreference.com/w/c/numeric/fenv)
```
#define VE_TONEARESTUP   /*implementation defined*/
#define VE_TONEARESTEVEN /*implementation defined*/
#define VE_DOWNWARD      /*implementation defined*/
#define VE_TOODD         /*implementation defined*/

int vegetxround();
int vesetxround(int round);
return 0 on success, non-zero otherwise.
```

zakk0610, 2020-08-25:
We had purposed  `vread_csr` and `vwrite_csr` interface [here](https://github.com/riscv/rvv-intrinsic-doc/blob/master/rvv-intrinsic-api.md#readwrite-urw-vector-csrs), but it is not included in intrinsic RFC and function list yet

```
enum RVV_CSR {
  RVV_VSTART = 0,
  RVV_VXSAT,
  RVV_VXRM,
  RVV_VCSR,
};

unsigned long vread_csr(enum RVV_CSR csr);
void vwrite_csr(enum RVV_CSR csr, unsigned long value);
```
Any comments or suggestions will be appreciated.

arcbbb, 2020-08-27:
I prefer HK's idea having specific API to read/set vector rounding mode.
In that way user don't need to know the format of CSR in order to read the content.

kito-cheng, 2020-08-27:
I prefer high-level API for that too, the only concern to me is it's not `intrinsic`.

HanKuanChen, 2020-09-30:
@kito-cheng do you think this is a semantic intrinsic?
If so, I think we should add this.

HanKuanChen, 2020-11-04:
I propose a new pull request https://github.com/riscv/rvv-intrinsic-doc/pull/46 and a simple implementation here.
```
#define VE_TONEARESTUP 0
#define VE_TONEARESTEVEN 1
#define VE_DOWNWARD 2
#define VE_TOODD 3
// Each of these macro constants expands to a nonnegative integer constant expression.

// return 0 on success, non-zero otherwise.
int vesetxround(int round)
{
	switch(round)
	{
	case VE_TONEARESTUP:
	case VE_TONEARESTEVEN:
	case VE_DOWNWARD:
	case VE_TOODD:
		asm volatile("csrw vxrm,%z0"::"rJ"(round));
		return 0;
	default:
		return -1;
	}
}

// return the current rounding macro, negative value if the rounding mode cannot be determined.
int vegetxround(void)
{
	int vxrm;
	asm volatile("csrr %0,vxrm":"=r"(vxrm));
	switch(vxrm)
	{
	case VE_TONEARESTUP:
	case VE_TONEARESTEVEN:
	case VE_DOWNWARD:
	case VE_TOODD:
		return vxrm;
	default:
		return -1;
	}
}

// clear vxsat
inline void veclearxsat(void)
{
	asm volatile("csrw vxsat,x0");
}

// return 1 if vxsat is set, 0 otherwise.
inline int vetestxsat(void)
{
	int vxsat;
	asm volatile("csrr %0,vxsat":"=r"(vxsat));
	return vxsat;
}
```

ebahapo, 2020-11-09:
Rather, shouldn't a library that supported RVV set `vxrm` as part of supporting `fenv.h` instead?

kito-cheng, 2022-08-01:
We define `vxrm` and `vxsat` as call clobber in psABI, so this proposal become impossible to implement now.

[1] https://github.com/riscv-non-isa/riscv-elf-psabi-doc/pull/294

closed 2022-08-01 by kito-cheng