riscv-non-isa/riscv-rvv-intrinsic-doc #37: Reinterpret between different LMUL under the same SEW

HanKuanChen, 2020-08-26:
A `vint8m1_t` type can definitely be changed to `vint8m8_t`.
A `vint8m8_t` type can be changed to `vint8m1_t` if `vl` is already known. (Of course, users should aware what they are doing.)
```
vint8m2_t vreinterpret_v_i8m1_i8m2 (vint8m1_t src);
vint8m4_t vreinterpret_v_i8m1_i8m4 (vint8m1_t src);
vint8m8_t vreinterpret_v_i8m1_i8m8 (vint8m1_t src);
...
vint8m1_t vreinterpret_v_i8m8_i8m1 (vint8m8_t src);
vint8m2_t vreinterpret_v_i8m8_i8m2 (vint8m8_t src);
vint8m4_t vreinterpret_v_i8m8_i8m4 (vint8m8_t src);
```

zakk0610, 2020-09-07:
`vint8m2_t` and `vint8m1_t` occupy the different size, I think it should not be called `reinterpret`.
Could you share the user scenario? 
Are you doing vector-length agnostic programming?
Or are you going to do the similar thing like https://github.com/riscv/rvv-intrinsic-doc/issues/28 mention, implement combine operation for vcombine instruction in NEON?

ChunyuLiao, 2020-12-28:
How to conversion between data types and mask types use rvv intrinsics?

Eg: <vscale x 1 x i64> <-> <vscale x 1 x i1>

zakk0610, 2020-12-28:
mask->data type:
`vuint64m1_t  data = vmerge_vxm_u64m1(mask, vundefined_u64m1(), 1);`

data type->mask:
`vbool64_t  mask = vmseq_vx_u64m1(data, 1);`

eopXD, 2022-07-29:
We have `vlmul_ext`, `vlmul_trunc` now. Closing the issue.

closed 2022-07-29 by eopXD