riscv-non-isa/riscv-rvv-intrinsic-doc #45: Support more util functions for vector status register.

JerryShih, 2020-11-04:
Currently, we have "vreadvl()" to get the vl, we could have more util functions for "mstatus", "vlenb", "vstart" and other status register.

arcbbb, 2020-11-09:
You could propose a simple implementation like https://github.com/riscv/rvv-intrinsic-doc/issues/36

eopXD, 2022-07-29:
Exposing `vlenb` may be a feature to consider, but there is no benefit to expose `mstatus` and `vstart`. Let us revisit this after v1.0.

eopXD, 2023-10-23:
Revisiting this issue after a year now...

`VLEN` is now exposed.

The `vstart` CSR is currently not exposed to the intrinsics programmer, and the intrinsics have the semantics of `vstart = 0`. So I think it would be strange for us to expose control to `vstart`. Additionally, support for positive `vstart` values is implementation -defined; thus, portable application software should not set `vstart > 0`.

Other than `vxsat` and `fflag` that is left to be exposed, I don't think `vstart` and `mstatus` would be our consideration. Please open up another issue if you have a strong demand to this feature.

closed 2023-10-23 by eopXD