riscv-non-isa/riscv-rvv-intrinsic-doc #47: Support the non-zero vstart value for vector load/store intrinsics.

JerryShih, 2020-11-05:
The vector load/store could support the non-zero vstart value.
Currently, the vector load/store intrinsic imply to call the vsetvl to update the EMUL. The vsetvl will reset the vstart to zero.

Thus, the following code will not work.
```
set_vstart(new_vstart_value)
dst=vload_e32_m1(ptr) // Since the vload implies a vsetvl to set the EMUL, the "new_vstart_value" will be cleared.

```

We might need to have the additional vstart parameter for the vector load/store intrinsic, and then set the new vstart value after the vsetvl instruction.

ebahapo, 2020-11-06:
Either that or the compiler will have to track the use of `set_vstart()`.  I'd love to drop `set_vstart()`, though. 😀

zakk0610, 2020-12-18:
IMHO, intrinsic should not support setting vstart function. 
RVV spec said that "The vstart CSR is writable by unprivileged code, but non-zero vstart values may cause vector instructions to run substantially slower on some implementations, `so vstart should not be used by application programmers.`"

I believe intrinsic users are application programmers too. :P

eopXD, 2022-07-29:
Looks like the discussion may have come a conclusion here. We suggest that advanced users that seeks to manipulate `vstart` to directly use the assembly.

closed 2022-07-29 by eopXD