riscv-non-isa/riscv-rvv-intrinsic-doc #48: Remove splats?

knightsifive, 2020-11-17:
The `v{f}splat` routines appear to just be simple aliases for the `v{,f}mv_v_{x,f}` routines. There's obviously no harm in including them; however, in such an enormous API I think we should take measures to reduce complexity whenever possible. Thus I propose removing the splats.

Less drastic measures are (1) to deprecate them now and revisit removing them sometime in the future, or (2) to make them an optional extension.

knightsifive, 2020-11-17:
Upon further study, it appears that all of the "semantic functions"
https://github.com/riscv/rvv-intrinsic-doc/blob/master/rvv_intrinsic_funcs/13_semantic_vector_functions.md
are redundant. Hmm. Obviously this was intentional, meaning that someone out there clearly wanted these aliases.

I guess these could all be implemented as macro wrappers around existing routines... Another idea is that we just formally define these routines as such, and provide a separate header file that implements these translations.

HanKuanChen, 2020-11-17:
I think the reason we have semantic functions is that the intrinsic document does not use [v-spec](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc) naming in the [beginning](https://github.com/riscv/rvv-intrinsic-doc/blob/2be63fc207f239f1d0c6691848f4d78523e6b985/rvv_intrinsic_funcs.md). Provide semantic functions help users programming easily.
However, after upgrading to [new interface](https://github.com/riscv/rvv-intrinsic-doc/blob/bb210940fa9b7c9c4e204c6cade7e901860349a0/rvv_intrinsic_funcs.md), the semantic functions somehow remain.

knightsifive, 2020-11-17:
Since they're vestigial and now redundant, I'm casting my vote in favor of excising them.

rednoah91, 2020-11-24:
From my using experience, I also vote for removing splats.

zakk0610, 2020-12-18:
It seems all we agree remove splat.
Another question is do we really need the "semantic functions"?
Personally I prefer to remove all "semantic functions" in intrinsic rfc.

zakk0610, 2020-12-30:
currently I remove splat function in https://github.com/riscv/rvv-intrinsic-doc/commit/17d4cb460cab39d1bb567af085ee493b74bd26b8.
Is ok to remove all "semantic functions" ?

HanKuanChen, 2020-12-30:
I think our opinion is remove whole semantic functions.

eopXD, 2022-07-29:
We don't have any "semantic functions" now. Closing this issue.

closed 2022-07-29 by eopXD