riscv-non-isa/riscv-rvv-intrinsic-doc #50: separate riscv_vector.h into riscv_vector.h and riscv_vector_vl.h

HanKuanChen, 2020-11-23:
I notice we use _riscv_vector.h_ for intrinsics with vl and without vl argument. May we riscv_vector.h into riscv_vector_vl.h and riscv_vector_vl.h?
If users wants to use intrinsics with vl argument, they could include _riscv_vector_vl.h_.
If users wants to use intrinsics without vl argument, they could include _riscv_vector.h_.
In addition, compilers may give a warning (or error) if users include these files in the same source.

kito-cheng, 2020-12-04:
Personally, I would prefer keep only one header for vector intrinsic including C11 generic one, but I know your point is we are not recommended mixed both implicitly and explicitly VL API together, so the having separated header and give error if included both could have a more strong impression to user that don't mix that.

But I think another point we are still debating is should we just keep one version? if so this issue would be gone :P but I guess it's hard to make decision at least within few weeks.

HanKuanChen, 2020-12-04:
We are not yet discussing which version should be remained. So current solution is provided two headers together for users.

However, we can still recommend users should use one style. If we get same opinion on which version should be used in the future, users can change intrinsic interface to another style easily.

eopXD, 2022-07-29:
We don't have intrinsics without VL now. Just to leave a record, the discussion on whether to include VL in the parameters are inside closed issue https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/8.

closed 2022-07-29 by eopXD