riscv-non-isa/riscv-rvv-intrinsic-doc #54: Support LMUL truncation and LMUL extension functions

zakk0610, 2020-12-18:
There are some requirements from users they want to keep current SEW but change LMUL (in https://github.com/riscv/rvv-intrinsic-doc/issues/28 and https://github.com/riscv/rvv-intrinsic-doc/issues/37)

maybe intrinsic functions could support LMUL truncation and LMUL extension regardless of vl.

ps.  I think this is not a `Reinterpret` operation because theirs VLEN are different.

naming could be `vlmul_[ext|trunc]_v_<src_type_with_lmul>_<dst_type_with_lmul>`

interfaces would looks like

```
// LMUL Extension, vlmul_ext_v_<src_lmul>_<target_lmul>
vint64m2_t vlmul_ext_v_i64m1_i64m2 (vint64m1_t op1);
vint64m4_t vlmul_ext_v_i64m1_i64m4 (vint64m1_t op1);
vint64m8_t vlmul_ext_v_i64m1_i64m8 (vint64m1_t op1);
vint64m4_t vlmul_ext_v_i64m2_i64m4 (vint64m2_t op1);
vint64m8_t vlmul_ext_v_i64m2_i64m8 (vint64m2_t op1);
vint64m8_t vlmul_ext_v_i64m4_i64m8 (vint64m4_t op1);

vint64m1_t vlmul_trunc_v_i64m2_i64m1 (vint64m2_t op1);
vint64m1_t vlmul_trunc_v_i64m4_i64m1 (vint64m4_t op1);
vint64m2_t vlmul_trunc_v_i64m4_i64m2 (vint64m4_t op1);
vint64m1_t vlmul_trunc_v_i64m8_i64m1 (vint64m8_t op1);
vint64m2_t vlmul_trunc_v_i64m8_i64m2 (vint64m8_t op1);
vint64m4_t vlmul_trunc_v_i64m8_i64m4 (vint64m8_t op1);
```

any thought?

zakk0610, 2020-12-29:
What do you think @joy2myself   @HanKuanChen?

HanKuanChen, 2020-12-29:
LGTM

joy2myself, 2020-12-29:
Hi @zakk0610 ,
Thanks for your proposal! But it doesn't seem like what I want in #28. I want to combine two (or more) small `LMUL` vectors to one big `LMUL` vector, or conversely, to split one big `LMUL` vector to two (or more) small `LMUL` vectors. Because my development is based on a fixed 128-bits `VLEN`, the `vl` should be changed with `LMUL`.

What I want may looks like
```
//vector combine
vint64m2_t vcombine_i64m1_i64m2 (vint64m1_t op1, vint64m1_t op2);
//vector split
vint64m1_t vget_low_i64m2_i64m1 (vint64m2_t op1);   //get the low half of big vector
vint64m1_t vget_high_i64m2_i64m1 (vint64m2_t op1);  //get the high half of big vector
```

In my opinion, two split functions can be implemented based on `slide` operation and `vl` modification, but I'm not sure how to implement combine operation. Moreover, even if these operations can be implemented by other intrinsics, will the interface packaged as such be more friendly to the front-end users?

HanKuanChen, 2020-12-29:
@joy2myself 
How about
```
vint64m2_t vcombine_i64m1_i64m2 (vint64m1_t op1, vint64m1_t op2, size_t vl)
{
    return vlmul_ext_v_i64m1_i64m2(vslideup(op1, op2, vl));
}
```

kito-cheng, 2020-12-29:
I guess it should `ext` before `slideup`, otherwise the result might truncated?

joy2myself, 2020-12-29:
Like this?
```
vint64m2_t vcombine_i64m1_i64m2 (vint64m1_t op1, vint64m1_t op2, size_t vl)
{
    return vslideup_vx_i64m2(vlmul_ext_v_i64m1_i64m2(op1), vlmul_ext_v_i64m1_i64m2(op2), vl);
}
```
I'm not sure does it work since `ext` operation modified `VLEN`, right?

zakk0610, 2020-12-29:
Those two utility functions would not modified vl so you need to reset vl if need.

like this would be better.
```
vint64m2_t vcombine_i64m1_i64m2 (vint64m1_t op1, vint64m1_t op2, size_t vl)
{
   vint64m2_t new_op1 = vlmul_ext_v_i64m1_i64m2(op1);
   vint64m2_t new_op2 = vlmul_ext_v_i64m1_i64m2(op2);
   size_t new_vl = vsetvl_e64m2 (vl)
   return vslideup_vx_i64m2(new_op1, new_op2, new_vl);
}
```

joy2myself, 2020-12-29:
Thanks! If it works, it looks good to me.

zakk0610, 2021-01-06:
I create a commit in https://github.com/riscv/rvv-intrinsic-doc/commit/28266f257c1d73fa33da74d33cc927ed639b98fd.
If there is no other questions, I will merge it.

closed 2021-02-04 by zakk0610