riscv-non-isa/riscv-rvv-intrinsic-doc #55: maskedoff argument to vcompress intrinsic might not be the best name

topperc, 2020-12-28:
Based on the behavior of the vcompress instruction, this argument seems to be the vector to use for tail elements. It doesn't have a direct relationship to the mask the way "maskedoff" does for other instructions. The lower popcnt(mask) elements of the result of this instrinsic come from "src" while the upper elements [vl-1:popcnt(mask)] elements come from "maskedoff". The position of the mask bits doesn't control which elements from maskedoff are used, rather the number of 1s in the mask does.

This also means that this intrinsic must use the tail undisturbed policy.

zakk0610, 2020-12-29:
Sorry for the confusion. I reuse the same pattern in doc generator.
So the interface should be  `vint8mf8_t vcompress_vm_i8mf8 (vint8mf8_t dest, vint8mf8_t src, vbool64_t filter);`
It means `vcompress` intrinsic does not require compiler should use the tail undisturbed policy.

zakk0610, 2020-12-29:
In fact, in intrinsic doc, we didn't specific tail policy clearly yet. 
I think before intrinsic start support user to select tail policy (second approach in https://github.com/riscv/rvv-intrinsic-doc/issues/27#issuecomment-649433549), maybe compiler implementation could generate `tu` policy as default for correctness first.

zakk0610, 2020-12-29:
I think we can rename it like `vint8mf8_t vcompress_vm_i8mf8 (vint8mf8_t dest, vint8mf8_t src, vbool64_t filter);`. 

What do you think?

topperc, 2020-12-29:
> I think we can rename it like `vint8mf8_t vcompress_vm_i8mf8 (vint8mf8_t dest, vint8mf8_t src, vbool64_t filter);`.
> 
> What do you think?

That looks good to me.

closed 2021-01-12 by topperc