riscv-non-isa/riscv-rvv-intrinsic-doc #58: Approximate reciprocal/rsqrt

jan-wassenberg, 2021-01-11:
It seems that vfrsqrte7 and vfrece7 from v0.9 are not yet added, perhaps pending the naming discussion in https://github.com/riscv/riscv-v-spec/issues/601 ?

zakk0610, 2021-01-12:
`vfrsqrte7` and `vfrece7` are not the 0.9 instruction.
https://github.com/riscv/riscv-v-spec/blob/0.9/v-spec.adoc

jan-wassenberg, 2021-01-12:
Agreed, they are in the [1.0 draft](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#added-reciprocal-and-reciprocal-square-root-estimate-instructions), sorry I was unclear (should have written "added since/changed from v0.9", not "from").

jan-wassenberg, 2021-01-12:
Ah, I see that this repo is currently targeting 0.9 only. Is the plan to update to 1.0 only after it is ratified?

zakk0610, 2021-01-12:
Sorry, what do you mean?
This repo has a [tag](https://github.com/riscv/rvv-intrinsic-doc/tree/v0.9) for v0.9 and master branch is targeting on 1.0 draft. 
You can find [`vfrsqrte7`](https://github.com/riscv/rvv-intrinsic-doc/blob/master/rvv_intrinsic_funcs/08_vector_floating-point_functions.md#vector-floating-point-reciprocal-square-root-estimate-functions) and [`vfrece7`](https://github.com/riscv/rvv-intrinsic-doc/blob/master/rvv_intrinsic_funcs/08_vector_floating-point_functions.md#vector-floating-point-reciprocal-estimate-functions) in master branch.

kito-cheng, 2021-01-12:
> Ah, I see that this repo is currently targeting 0.9 only. Is the plan to update to 1.0 only after it is ratified?

Yeah, we are plan to upgrade that to 1.0 once it ratified or tagged, but the time is much later than our expect...

kito-cheng, 2021-01-12:
Oh, I forgot @zakk0610 has been update before :P

kito-cheng, 2021-01-12:
Seems like we forgot to update the doc about the version info :P
https://github.com/riscv/rvv-intrinsic-doc/pull/59

zakk0610, 2021-01-12:
> Ah, I see that this repo is currently targeting 0.9 only. Is the plan to update to 1.0 only after it is ratified?

oh, sorry. I forget to update version info.

jan-wassenberg, 2021-01-12:
Thank you @kito-cheng and @zakk0610  for clarifying the version number and plan :)

eopXD, 2022-07-29:
The ratified v-spec v1.0 don't include these instructions. This issue is no longer valid.

closed 2022-07-29 by eopXD