riscv-non-isa/riscv-rvv-intrinsic-doc #61: Casting vbool to vector

jan-wassenberg, 2021-01-19:
Hi, storing mask bits seems like a reasonable thing to want to do.

In assembly we can just use the v0 register, but in intrinsics vbool and vuint are separate types and I haven't seen any way to convert them.
```
  vuint8m8_t v = vmv_v_x_u8m8(0);
  vbool1_t m = vmseq_vv_u8m8_b1(v, v);
  uint8_t bits[256];
  vse8_v_u8m1(bits, m);
```
error: cannot convert 'vbool1_t' to 'vuint8m1_t'

Is there a way to do this, or can we add one?

knightsifive, 2021-01-19:
Hi Jan,

First, let me mention that there is an active proposal in the V-extension task group to add mask load and store instructions. This is motivated by reducing costs in certain OOO implementations. It looks like the latest meeting minutes haven't been uploaded, but it was also discussed last month ([minutes](https://github.com/riscv/riscv-v-spec/blob/master/minutes/20201218-V-minutes.txt#L24)). Reading the tea leaves, this will become part of RVV 1.0, thence we will add a new intrinsic.

However, I think I agree with you that we should extend the [`vreinterpret` intrinsics](https://github.com/riscv/rvv-intrinsic-doc/blob/master/rvv_intrinsic_funcs_vl/12_miscellaneous_vector_functions.md#reinterpret-cast-conversion-functions) to convert `vbool` to other types.

I hope others can chime in.

jan-wassenberg, 2021-01-19:
Hi Nick, thanks for sharing. Both sound good to me.

kito-cheng, 2021-01-19:
OpenCV folks has ask for similar feature too, I think `vle1.v`/`vse1.v` would be better than  reinterpret, since it's hard (or error prone) to set the `VL` for load/store a masking via `vle8`/`vse8`, since actually you only need read/write `VL`/8 bytes.

knightsifive, 2021-01-19:
@kito-cheng how can I reinterpret a `vbool` as another type using the intrinsics API? 

(I don't have an application in mind, just curious.)

kito-cheng, 2021-01-19:
Hmmm, need waste some space and instruction to do that:
```c
uint8_t preds[256] = {0, 1, 0, 1...};
vuint8m8_t v = vle8_v_u8m1(preds);
vbool1_t mask = vmsne_vx_u8m8_b1 (v, 0);
```

kito-cheng, 2021-01-19:
I know it's not reinterpret, but we don't have reinterpret between `m1` <-> `vbool` yet :P
that's the solution we suggest to other guys who want pre-computed mask.

jan-wassenberg, 2021-01-19:
:) We're also [using vid_v](https://github.com/google/highway/blob/master/hwy/ops/rvv-inl.h#L1302) for a similar purpose.

zhaochenyuangit, 2022-03-09:
The riscv-v-spec 1.0 mentions a trick regarding to your request in Section 11.16 : Mask values can be widened into SEW-width elements using a sequence vmv.v.i vd, 0; vmerge.vim vd, vd, 1, v0.
With the latest v-spec and explicit VL intrinsics, the following code in C works for me:
  enum {AVL=20,};
  vuint8m8_t v = vmv_v_x_u8m8(0, AVL);
  vbool1_t mask = vmseq_vv_u8m8_b1(v, v, AVL);
  uint8_t one = 1;
  vuint8m8_t extended_mask = vmerge_vxm_u8m8(mask, v, one, AVL);
  uint8_t bits[256];
  vse8_v_u8m8(bits, extended_mask, AVL);

eopXD, 2022-08-02:
We now have `vsm_v` for saving boolean value into scalar.
Let's open a new issue if there is any issue with the existing API. Closing this one for now. Thank you :)

closed 2022-08-02 by eopXD