riscv-non-isa/riscv-rvv-intrinsic-doc #65: New generic intrinsic function rule.

zakk0610, 2021-02-04:
I would like to purpose a different overloading rule to reduce the number of intrinsic functions when using generic API.

1. We could use the approach mention in https://github.com/riscv/rvv-intrinsic-doc/issues/21#issuecomment-638635766  (ex. use `overloadable` attribute in clang) instead of C11 generic selection to support overload functions with different number of arguments.
2. Support overloaded as more as possible.  Consider function arguments to decide unsupported instructions. 
   1. scalar arguments alone. (non-masked vle/vlse, etc.)
   2. empty function argument. (vmclr.m/vmset.m/vid.v)
   3. boolean vector arguments with return type of a non boolean vector. (viota)

Applying above rules we could reduce ~78% number of intrinsic functions when using overloading API.  (from 13106  to 2716)

Any suggestions?

The detail of changed  shows below:

### Unsupported overloaded functions:
 1. non-masked `vle`/`vlse` and related segment load instructions.
```
vint8m1_t vle8_v_i8m1 (const int8_t *base);
vint8m2_t vle8_v_i8m2 (const int8_t *base);
```
 2. non-masked `viota.m`
```
vuint8mf8_t viota_m_u8mf8 (vbool64_t op1);
vuint16mf4_t viota_m_u16mf4 (vbool64_t op1);​
```
 3.  `vmv.v.x`, `vfmv.v.f`
```
vint8m1_t vmv_v_x_i8m1 (int8_t src);
vint8m2_t vmv_v_x_i8m2 (int8_t src);
```
 4. `vmclr.m`/`vmset.m`/`vid.v` (empty argument)

### Supported new overloaded functions:
1. masked `vle`/`vlse` and related segment load instructions.
2. `vse`, `vsse` and indexed load/store and related segment load/store instructions.
3. masked `viota.m`
4. one overloading function for `vmv.v.v`, `vmv.x.s`, `vmv.s.x`, `vfmv.f.s`, `vfmv.s.f`
5. one overloading function for `vmadc.vvm`, `vmadc.vxm`, `vmadc.vv`, ` vmadc.vx`
6. one overloading function for `vmsbc.vvm`, `vmsbc.vxm`, `vmsbc.vv`, `vmsbc.vx`

kito-cheng, 2021-02-05:
The proposal is LGTM, the reason for those function didn't overloading is because the `symmetric` issue before, I thought load/sotre intrinsic should be `symmetric` would be more elegant, but the real situation is: without those overloading the number of intrinsic function is really...amazing.

So I am support this proposal.

zakk0610, 2021-02-17:
It seems mask and non masked function also can be one function.
ex. `vadd` for `vadd_vv_i8mf8 ` and `vadd_vv_i8mf8_m`

zakk0610, 2021-02-19:
PR https://github.com/riscv/rvv-intrinsic-doc/pull/67 for this issue.

closed 2021-03-11 by zakk0610