riscv-non-isa/riscv-rvv-intrinsic-doc #68: Use wider integers for shifts

knightsifive, 2021-02-24:
I'm concerned about several intrinsics which input a "shift" amount as `uint8_t`. This only allows shifts up to 255, which fails to expose underlying functionality for larger SEW (and ELEN).

Particular examples are `vsll_vx`, `vsr{a,l}_vx`, `vnsr{a,l}_vx`, `vssr{a,l}_vx`, and `vnclip{,u}_wx`. I only skimmed quickly; perhaps there are more.

The `vv`/`wv` forms use an appropriate type-width --- one that increases with SEW --- for their (vector) `op2`, and I suggest matching that in the `vx`/`wx` forms.

(It's true that the `vi` forms are restricted to a 5-bit `op2`, but these forms aren't exposed in the intrinsics API.)

zakk0610, 2021-02-25:
Agree, spec already reserves SEW=1024 supported.

Unfortunately `uintXLEN_t` is still in the PR, https://github.com/riscv/riscv-c-api-doc/pull/14
Which type are you prefer to use? `unsigned long`?

knightsifive, 2021-02-25:
I propose we just define `uint_xlen_t` in the usual way in the `riscv_vector.h`, and delete it when the C API PR is approved and added to compilers. Hopefully this can be done without breaking user code because `riscv_vector.h` is bundled with the toolchain.

I'm pretty sure the P-extension does the same thing in their intrinsics header file. I don't know about other extensions' intrinsics.

I'm not an expert on this kind of thing, happy to hear alternatives.

kito-cheng, 2021-02-25:
> I propose we just define uint_xlen_t in the usual way in the riscv_vector.h, and delete it when the C API PR is approved and added to compilers. Hopefully this can be done without breaking user code because riscv_vector.h is bundled with the toolchain.

That SGTM, I think intrinsic for B extension should need that too.

zakk0610, 2021-03-11:
Before we have `uint_xlen_t`, could we use `size_t`?

nick-knight, 2022-03-03:
This was addressed by #72 , although I think this and related aspects of the API should be revisited once we have XLEN-width integer types to replace `size_t`, `long`, etc.

closed 2022-03-03 by nick-knight