riscv-non-isa/riscv-rvv-intrinsic-doc #69: remove vbool_t except vbool1_t

HanKuanChen, 2021-02-25:
MLEN is 1 since [0.9](https://github.com/riscv/riscv-v-spec/blob/0.9/v-spec.adoc#sec-mask-register-layout) spec.

Is
```
vbool2_t
vbool4_t
vbool8_t
vbool16_t
vbool32_t
vbool64_t
```
still be required?

jan-wassenberg, 2021-02-25:
I think these could be useful guardrails to prevent accidentally mixing masks for u8m8 (all of v0) vs u16m8 (half of v0).

These could also be useful to compute VL when using vse8 to store a mask (assuming we are later able to cast from vbool to vector: https://github.com/riscv/rvv-intrinsic-doc/issues/61).

Does that make sense?

HanKuanChen, 2021-02-25:
Before 0.9 spec, the `vbool_t` type is designed to encode `MLEN` (which is `sew/lmul`) in type. It makes sense. Because they occupy different widths in a masked register. And masked vector instructions also examine specified bits in a masked register.
e.g.,
A sew 32 and lmul 4 masked vector operation see `mask[0]`, `mask[8]`, `mask[16]` ...

Users use a `vbool1_t` for a sew 32 and lmul 4 masked vector operation is wrong.

However, since 0.9 spec, MLEN is modified to 1. In addition, `vle1.v` and `vse1.v` are presented.
If we still want to use `vbool2_t` to `vboot64_t`, the intrinsics have to be designed to
```
vbool1_t vle1_v_b1 (const uint8_t *base, size_t vl);
vbool2_t vle1_v_b2 (const uint8_t *base, size_t vl);
vbool4_t vle1_v_b4 (const uint8_t *base, size_t vl);
vbool8_t vle1_v_b8 (const uint8_t *base, size_t vl);
vbool16_t vle1_v_b16 (const uint8_t *base, size_t vl);
vbool32_t vle1_v_b32 (const uint8_t *base, size_t vl);
vbool64_t vle1_v_b64 (const uint8_t *base, size_t vl);
void vse1_v_b1 (const uint8_t *base, vbool1_t value, size_t vl);
void vse1_v_b2 (const uint8_t *base, vbool2_t value, size_t vl);
void vse1_v_b4 (const uint8_t *base, vbool4_t value, size_t vl);
void vse1_v_b8 (const uint8_t *base, vbool8_t value, size_t vl);
void vse1_v_b16 (const uint8_t *base, vbool16_t value, size_t vl);
void vse1_v_b32 (const uint8_t *base, vbool32_t value, size_t vl);
void vse1_v_b64 (const uint8_t *base, vbool64_t value, size_t vl);

vbool1_t vmand_mm_b1 (vbool1_t op1, vbool1_t op2, size_t vl);
vbool2_t vmand_mm_b2 (vbool2_t op1, vbool2_t op2, size_t vl);
vbool4_t vmand_mm_b4 (vbool4_t op1, vbool4_t op2, size_t vl);
vbool8_t vmand_mm_b8 (vbool8_t op1, vbool8_t op2, size_t vl);
vbool16_t vmand_mm_b16 (vbool16_t op1, vbool16_t op2, size_t vl);
vbool32_t vmand_mm_b32 (vbool32_t op1, vbool32_t op2, size_t vl);
vbool64_t vmand_mm_b64 (vbool64_t op1, vbool64_t op2, size_t vl);
...
```
instead of
```
vbool1_t vle1_v (const uint8_t *base, size_t vl);
void vse1_v (const uint8_t *base, vbool1_t value, size_t vl);

vbool1_t vmand_mm (vbool1_t op1, vbool1_t op2, size_t vl);
..
```

> I think these could be useful guardrails to prevent accidentally mixing masks for u8m8 (all of v0) vs u16m8 (half of v0).

The length of the actual bits stored in a masked register is depended on `vl`. Not on `sew/lmul`.

jan-wassenberg, 2021-02-25:
I agree vse1.v (when available in compilers) will avoid most vse8 for masks.

> The length of the actual bits stored in a masked register is depended on vl. Not on sew/lmul.

Perhaps I am misunderstanding something. Let's imagine we call vmsle_vv_i16m8_b2 and get a vbool2_t, then pass that to vmerge_vxm_i8m8.
Oops, this may be an error: the argument should have been vbool1_t, with vbool2_t the upper half of the bits are probably invalid.
Does it make sense to have this be a compile error to prevent such accidents?

HanKuanChen, 2021-02-26:
> Oops, this may be an error: the argument should have been vbool1_t, with vbool2_t the upper half of the bits are probably invalid.

No. MLEN is 1 now. They only difference is the type name. Vector instructions examine `mask[0]` for the first element, `mask[1]` for the second element, etc.

> Does it make sense to have this be a compile error to prevent such accidents?

Indeed, have other `vbool` types can sometimes prevent users passing a wrong `vbool` type.

However, `vlmul_ext` and `vlmul_trunc` are also presented. That means users may use a different lmul vector instruction but keep using the same mask. But we don't provide `vbool` convert intrinsics here.

kito-cheng, 2021-02-26:
> However, vlmul_ext and vlmul_trunc are also presented. That means users may use a different lmul vector instruction but keep using the same mask. But we don't provide vbool convert intrinsics here.

I think it's different thing, yes, we provided those API for danger operation to break the type safe, but this way need user explicitly to write down what they want to do.

And back to the `vbool*_t` case, remove the the number from the type it's kind of removing security lock by default, so my view is: vlmul_ext and vlmul_trunc is give user a way to removing security lock, but not by default.

This also remind me the story about C++, why reinterpret_cast named so ugly, it's intentionally to make sure user aware they are doing ugly and danger things, IIRC Bjarne has write down the reason in his book, but I don't have one on my office right now, so here is only a interview with Bjarne has mention this a little bit[1] :P

[1] https://www.stroustrup.com/devXinterview.html

kito-cheng, 2021-02-26:
> But we don't provide vbool convert intrinsics here.
Oh I miss this sentence, I think it's good idea to provide this.

kito-cheng, 2021-02-26:
Just for reference, I remember @Hsiangkai @zakk0610 and me have discussion on that when MLEN is removed, and Kai has update the doc to remove the term `MLEN` from the document at that moment.

This is corresponding change:
https://github.com/riscv/rvv-intrinsic-doc/commit/cd9cf2c74674d072a453fa5887dc0081d0d4e18e#diff-6c2ab3f40423a568fd35af88809cbd79829c3c62168a6e93b78a4f4183633f8bR64

jan-wassenberg, 2021-02-26:
> Vector instructions examine mask[0] for the first element, mask[1] for the second element, etc.

Right, my concern is that we might have user code that sets only mask[0..VL/2 - 1], and that
without type safety (different vbool##_t), we pass it to something that expects mask[0..VL - 1] to be valid.

> I think it's good idea to provide this [vbool convert intrinsics].

+1, this sounds useful - have the 'security lock' by default, but also allow turning it off.

eopXD, 2022-08-02:
Adding these functionality is rather simple and don't need much implementation. I have just opened an RFC thread #156 to gather use cases and try to come up a proposal in this v1.0 release. Closing this issue and let's discuss further there. Thank you for the issue!

closed 2022-08-02 by eopXD