riscv-non-isa/riscv-rvv-intrinsic-doc #73: overloading and non-overloading header file organization

zakk0610, 2021-03-24:
When users want to use overloading api, do they need to change the include header? (for example, use riscv_vector_overloaded.h instead of riscv_vector.h)

I prefer the toolchain provide one header `riscv_vector.h` for users.

The benefit is keeping simple and users don't need to change header.
But the drawback is it would increase the size of non-overloading header.

kito-cheng, 2021-03-24:
I would prefer keep only one header too, I think compilation speed could be improved by PCH or [pragma trick which used by AArch64 GCC](https://github.com/gcc-mirror/gcc/blob/master/gcc/config/aarch64/arm_sve.h#L40)

ChunyuLiao, 2021-03-25:
If we only keep one header file(riscv_vector.h), do we need to emit riscv_vector_overloaded.h?

zakk0610, 2021-03-25:
> If we only keep one header file(riscv_vector.h), do we need to emit riscv_vector_overloaded.h?

@ChunyuLiao 
I think it's about implementation, so I updated my patch, thanks!

closed 2021-07-02 by zakk0610