riscv-non-isa/riscv-rvv-intrinsic-doc #74: Utility of vmmv.m and vmv.v.v intrinsics

topperc, 2021-03-24:
Shouldn't users be encouraged to use operator= to copy vectors like any other variable? Using a mv intrinsic guarantees an instruction is emitted by the compiler even if register allocation would allow the input and output to use the same register. Unless we teach the compiler to look for these operations and turn them into copies that register allocation can coalesce.

HanKuanChen, 2021-03-25:
`operator=` is ambiguity. Users may want to copy a vector with only `vl` elements, while others may want to copy a vector with `vlmax` elements (when tail is undisturbed).

zakk0610, 2021-03-25:
I'm not sure the `operator=` behavior is equal to `vmv.v.v` and `vmmv.m`.
[Some ](https://github.com/riscv/rvv-intrinsic-doc/issues/13#issuecomment-792582941)people think `operator=` is VLMAX operation and mv intrinsics operate on `vl` element.
If they are different. I think the mv intrinsics are still useful.

topperc, 2021-03-25:
Wouldn't the vmv.v.v intrinsic need an extra operand for the undisturbed values to be useful for tail undisturbed? Right now the register allocator will just use whatever register is convenient as the destination so you have no control. At best it's a performance optimization to only copy VL elements.

kito-cheng, 2021-03-25:
I would suggest `operator=` should have `VLMAX` semantic and then compiler could optimize to `vmv.v.v` if possible, user should use intrinsic if user only want to copy `vl`.

So I agree with @topperc, we should encourage user to use `operator=`.

nick-knight, 2021-03-25:
What happens when using operator= on a type with fractional LMUL?

topperc, 2021-03-25:
> What happens when using operator= on a type with fractional LMUL?

I think it becomes a vmv1r.v if a copy is needed.

topperc, 2021-03-25:
I forgot about VL when I started this issue, but there's been some good discussion here.

Using vmv.v.v can improve performance by limiting the number of elements copied. But forcing users to explicitly code all copies using intrinsics to get the best performance seems bad. Variables in C should be a different abstraction than hardware registers.

To handle the tail undisturbed case we need a new intrinsic that takes that undisturbed values as a second input. This would not apply to vmmv.m since I mask instructions are always tail agnostic if I'm reading the spec correctly.

eopXD, 2022-08-02:
We currently have `=` with semantic of a [whole vector register move](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#166-whole-vector-register-move) (which is also `vmv.v.v` of VL=VLMAX).

We should explicitly emphasize this in the document of v1.0 since the assumption is necessary for redundant move eliminations.

PR #287 (2023-10-25): Add more details in the specification
While cleaning up the issue threads, I found some more details we need to have in the specification.

This PR:
- Resolves #74 by adding a description of "Semantic of assignment (`=`) in the C language" under the programming note section of the spec
- Resolves #242 by adding a description of "Mixing inline assembly and intrinsics" under the programming note section of the spec
- Replace `float16_t` with `_Float16` that is now supported by both LLVM and GCC.  This keeps us away from the need to define our own type. (Resolves #31, and alongside also resolves  [... cut]
closed 2023-10-25 by eopXD