riscv-non-isa/riscv-rvv-intrinsic-doc #78: Document that strided load/store with stride of 0 can be optimized to X0.

topperc, 2021-04-20:
The V extension spec mentions

> When rs2=x0, then an implementation is allowed, but not required, to perform fewer memory operations than the number of active elements, and may perform different numbers of memory operations across different dynamic executions of the same static instruction.
> 
> Note | Compilers must be aware to not use the x0 form for rs2 when the immediate stride is 0 if the intent to is to require all memory accesses are performed.

We should note in the intrinsic spec that if the compiler determines the stride is 0, that a compiler implementation is allowed to use X0 as the stride which will not perform all memory accesses. If the user needs all memory accesses to be performed, they can use an indexed load/store with all zero indices.

The alternative is to define the intrinsic to always put 0 in a register and have a new intrinsic for the X0 behavior. Since needing all memory accesses isn't likely to be common, I don't think we shouldn't increase intrinsic complexity to support it.

kito-cheng, 2021-04-21:
I think it should not common to require all memory accesses are performed, so optimize 0 stride to `x0` and add note for that is reasonable to me.

kito-cheng, 2021-04-21:
@rofirrim @rdolbeau what do you think? does the optimize make sense and safe to you ?

rofirrim, 2021-04-21:
Totally reasonable to me.

zakk0610, 2021-04-21:
I think the we could write a programming note to suggest user should use indexed load/store with all zero indices for all memory access. In other word, using the strided load/store with stride of 0, compiler does not guarantee to generate an all memory accesses instruction.

zakk0610, 2021-04-21:
I made https://github.com/riscv/rvv-intrinsic-doc/pull/80 for this issue.

closed 2021-05-25 by zakk0610