riscv-non-isa/riscv-rvv-intrinsic-doc #83: RISC-V Vector Intrinsics types missing casts between same size vectors and groups of vectors

tony-cole, 2021-04-29:
Originally posted on: https://github.com/riscv/riscv-gcc/issues/259 but it was suggested I post on here.

Re: riscv32-unknown-elf-gcc (GCC) 10.1.0 (intrinsic-patch)

I can't find a way to cast vfloat32m4_t to vfloat32m2x2_t and visa-versa. Please let me know if this is currently possible or not.

I'm trying to:

load vfloat32m4_t
extract the lower vfloat32m2_t and use that for processing
slide down the vfloat32m4_t
load the upper vfloat32m2_t of the vfloat32m4_t
repeat, etc
I found the tuple and segment load/store intrinsics which almost do what I want, but I can't cast a vfloat32m4_t to vfloat32m2x2_t and visa-versa, GCC gives an error if I just use the normal C casting method.

I need GCC to accept explicit C casts between equivalent sized vector types, or add more intrinsics to do the same, something like:

vfloat32m2x2_t   vcast_f32m4_f32m2x2 (vfloat32m4_t   src);
vfloat32m4_t     vcast_f32m2x2_f32m4 (vfloat32m2x2_t src);
and of course all the other type combinations that have the same register (LMUL) sizes.

For me the explicit C casting method is preferable (implicit casting should still generate an error as this is very useful for catching mistakes).

There will be many other uses, for instance casting vuint8m8_t to vint8m8_t or to any of the other types (e.g. vuint32m8_t).

jan-wassenberg, 2021-04-30:
@tony-cole lmul trunc/ext were recently added and can be useful for upper/lower half (with slidedown):
https://github.com/riscv/rvv-intrinsic-doc/pull/57 https://github.com/riscv/rvv-intrinsic-doc/issues/54
Does that cover your use case?

tony-cole, 2021-05-04:
Thanks, but not quite.

trunc will allow me to extract the lower vfloat32m2_t (of a vfloat32m4_t) and use that for processing, but I want to use it in-place as, I think, trunc copies it to another register? Which defeats the optimisations I'm trying to make.

Also, I can't see a way to load data directly to the upper vfloat32m2_t of a vfloat32m4_t. 

Basically I want ways to access in-place (without copying to other registers) smaller parts of (or groups within) a Vector register (LMUL) group (on the correct boundaries of course). This would allow me to optimise my code further.

In asm I could do it, but the intrinsics don't allow me to (as far as I can see).

jan-wassenberg, 2021-05-04:
It seems feasible for a compiler to understand that trunc of an m4 are the same registers as its lower half, but recombining currently requires a vslideup op which I suspect is harder to optimize into a no-op (by allocating adjacent registers).
Maybe someone with knowledge of the compiler can comment?

zakk0610, 2021-05-04:
truncation could be in-place operation.
please see the example https://godbolt.org/z/5nY7sEcnW

tony-cole, 2021-05-07:
Thanks.

I have discovered the GCC compiler switch: "-flax-vector-conversions" which allow me to cast between vector types that are the same size, e.g. vint32m4_t <-> vint64m4_t. 

But GCC still generates an incompatible type error when coverting vfloat32m2x2_t <-> vfloat32m4_t, even though they are the same size. As long as my assumption is correct in that tuple types are consecutive and a simple cast should be possible for the same sized vectors?

eopXD, 2022-08-02:
This is a request that is related to sizeless structures and its implementation. Maybe we can hold this for discussion after the v1.0 release?

closed 2022-08-02 by eopXD