riscv-non-isa/riscv-rvv-intrinsic-doc #84: CSR register Vector rounding mode #defines missing

tony-cole, 2021-04-29:
This was originally posted on: https://github.com/riscv/riscv-gcc/issues/256

I was expecting to find some #defines for the rounding modes in riscv-vector.h, something like:

/* Vector Fixed-Point Rounding Mode Register vxrm settings
Use with vwrite_csr(RVV_VXRM, RVV_VXRM_XXX) */

#define RVV_VXRM_RNU (0) /* Round-to-nearest-up (add 0.5 LSB) /
#define RVV_VXRM_RNE (1) / Round-to-nearest-even /
#define RVV_VXRM_RDN (2) / Round-down (truncate) /
#define RVV_VXRM_ROD (3) / Round-to-add (OR bits into LSB, aka "jam") */

Could use double underscore between field and mode value, depending upon your convention, e.g.:
#define RVV_VXRM__RNU (0) /* Round-to-nearest-up (add 0.5 LSB) /
#define RVV_VXRM__RNE (1) / Round-en to-nearest-even /
#define RVV_VXRM__RDN (2) / Round-down (truncate) /
#define RVV_VXRM__ROD (3) / Round-to-add (OR bits into LSB, aka "jam") */

jan-wassenberg, 2021-04-30:
Somewhat related: do we plan to add intrinsics for setting FRM? Currently we use fsrm in asm.

tony-cole, 2021-05-04:
In the riscv_vector.h file (in gcc rv32gc 10.1.0 rvv-intrinsic-patch) I use vread_csr() and vwrite_csr() intrinsics:

```
// Code from riscv_vector.h

enum RVV_CSR {
  RVV_VSTART = 0,
  RVV_VXSAT,
  RVV_VXRM,
  RVV_VCSR,
};

__extension__ extern __inline
__attribute__ ((__always_inline__, __gnu_inline__, __artificial__))
unsigned long vread_csr(enum RVV_CSR csr)
{
  unsigned long rv = 0;
  switch (csr)
    {
    case RVV_VSTART:
      __asm__ __volatile__ ("csrr\t%0,vstart" : "=r"(rv) : : "memory");
      break;
    case RVV_VXSAT:
      __asm__ __volatile__ ("csrr\t%0,vxsat" : "=r"(rv) : : "memory");
      break;
    case RVV_VXRM:
      __asm__ __volatile__ ("csrr\t%0,vxrm" : "=r"(rv) : : "memory");
      break;
    case RVV_VCSR:
      __asm__ __volatile__ ("csrr\t%0,vcsr" : "=r"(rv) : : "memory");
      break;
    }
  return rv;
}

__extension__ extern __inline
__attribute__ ((__always_inline__, __gnu_inline__, __artificial__))
void vwrite_csr(enum RVV_CSR csr, unsigned long value)
{
  switch (csr)
    {
    case RVV_VSTART:
      __asm__ __volatile__ ("csrw\tvstart,%z0" : : "rJ"(value) : "memory");
      break;
    case RVV_VXSAT:
      __asm__ __volatile__ ("csrw\tvxsat,%z0" : : "rJ"(value) : "memory");
      break;
    case RVV_VXRM:
      __asm__ __volatile__ ("csrw\tvxrm,%z0" : : "rJ"(value) : "memory");
      break;
    case RVV_VCSR:
      __asm__ __volatile__ ("csrw\tvcsr,%z0" : : "rJ"(value) : "memory");
      break;
    }
}
```

It would be useful if the register values/bit-fields/settings were also defined here for RVV_VXRM and the other CSR registers.

HanKuanChen, 2021-06-30:
I think you need to this https://github.com/riscv/rvv-intrinsic-doc/pull/46?
@Hsiangkai

eopXD, 2022-08-02:
Both LLVM and GCC have implemented these API. This issue is resolved now.

closed 2022-08-02 by eopXD