riscv-non-isa/riscv-rvv-intrinsic-doc #86: Missing changes for "Widening Vector-Scalar Arithmetic Instructions."

pshung, 2021-05-21:
1. In document overloaded_intrinsic_funcs.md,  vwadd should be named vwadd_vv.
2.  Does vwmul should follow the same naming rule?

pshung, 2021-05-21:
2. Does vwmul should follow the same naming rule?  If yes, all APIs should be changed.
3. Does vwmacc_vx should follow?

zakk0610, 2021-05-21:
> 1. In document overloaded_intrinsic_funcs.md, vwadd should be named vwadd_vv.

Thanks. I added them back cause by 00882f1.

>2. Does vwmul should follow the same naming rule?

In original design this naming rule only apply to widening vector-scalar arithmetic if they have `wx` instructions.
I will update RFC to make it more clear, thanks.

zakk0610, 2021-05-21:
https://github.com/riscv/rvv-intrinsic-doc/pull/88

closed 2021-06-02 by pshung