riscv-non-isa/riscv-rvv-intrinsic-doc #92: overloaded reinterpreter API

pshung, 2021-05-23:
The content shows the low-level API in overloaded_intrinsic_funcs.md rather than the overloaded API.

zakk0610, 2021-05-25:
Sorry, the title is confusing. Do you mean low-level API should not show in overloaded_intrinsic_funcs.md?

pshung, 2021-05-25:
Now, it shows the low-level functions
 vint16m1_t vreinterpret_v_f16m1_i16m1(vfloat16m1_t src);
rather than
vint16m1_t vreinterpret_i16(vfloat16m1_t src);

pshung, 2021-06-02:
BTW,

Overloaded APIs for vreinterpret are not generated in the following commit.
[main] 8f683366afcfd7c03faea615d1529a0014d7dbde - commit 4248 of 389948                                   1%
commit 8f683366afcfd7c03faea615d1529a0014d7dbde
Refs: llvmorg-13-init-7485-g8f683366afcf
Author:     Zakk Chen <zakk.chen@sifive.com>
AuthorDate: Mon Apr 12 19:01:44 2021 -0700
Commit:     Zakk Chen <zakk.chen@sifive.com>
CommitDate: Fri Apr 16 09:41:19 2021 -0700

    [RISCV][Clang] Add RVV miscellaneous intrinsic functions.

    1. vreinterpret
    2. vundefined
    3. LMUL truncation and extension.

    Reviewed By: craig.topper

    Differential Revision: https://reviews.llvm.org/D100391

zakk0610, 2021-06-02:
Yes, currently overloaded document also shows the low-level APIs.
I think it's more convenient for users, they can easily understand  which intrinsic functions have no overloading supported.

so maybe they can understand that all miscellaneous intrinsic functions have no overloading supported now.
maybe we need to add more description for that in somewhere.

pshung, 2021-06-02:
I may misunderstand your comment.
Do you mean this document won't show those unimplemented instructions?
If yes, why this document shows all unimplemented pseudo vector instructions? such as 'vfabs' ?

pshung, 2021-06-02:
Do you have any plan to implement pseudo vector instructions at the intrinsic level?

vwcvtu.x.x.v, vnot.v, vneg.v, vncvt.x.x.w, vmsgeu.vx, vmsgeu.vv, vmset.m, vmnot.m, vmmv.m, vmclr.m, vfneg.v, vfabs.v

Those instructions are documented, but not implemented on LLVM yet.

zakk0610, 2021-06-02:
1. I believe LLVM trunk had supported all pseudo vector instructions.
please see https://godbolt.org/z/onfzPv3cE.

2. about zfh, we need have more discussion about which type we should use for fp16:
https://github.com/riscv/rvv-intrinsic-doc/issues/18#issuecomment-818472454

3. about the overloading vreinterpret, I will update the document and support overloading api later.
Thanks for point out.

pshung, 2021-06-02:
Thanks for all feedback.

closed 2021-06-02 by pshung