riscv-non-isa/riscv-rvv-intrinsic-doc #94: Reinterpret between different LMUL and different SEW

rjiejie, 2021-05-31:
case from OpenCV:

inline v_uint8x16 v_pack_b(const v_uint32x4& a, const v_uint32x4& b, const v_uint32x4& c, const v_uint32x4& d)
{
        unsigned CV_DECL_ALIGNED(32) ptr[16] = {0};
        v_store(ptr, a);
        v_store(ptr + 4, b);
        v_store(ptr + 8, c);
        v_store(ptr + 12, d);
        vsetvlmax_e8m1();
        return v_uint8x16(vnsrl_wx_u8m1(vnsrl_wx_u16m2(vle32_v_u32m4(ptr), 0), 0));
}

The issue is that memory buffer is needed by changing v_uint32x4 to v_uint8x16.

to fix this memory issue, we designed the some intrinsics "vset_u32m4" like "vector_tuple_types":

inline v_uint8x16 v_pack_b(const v_uint32x4& a, const v_uint32x4& b,
                           const v_uint32x4& c, const v_uint32x4& d)
{
    vuint32m4_t vabcd = vundefined_u32m4();    \
    vuint16m2_t v16 = vundefined_u16m2();    \
    vabcd = vset_u32m4(vabcd, 0, a.val);    \
    vabcd = vset_u32m4(vabcd, 1, b.val);    \
    vabcd = vset_u32m4(vabcd, 2, c.val);    \
    vabcd = vset_u32m4(vabcd, 3, d.val);    \
    v16 = vnsrl_vx_u16m2(vabcd, 0, 16);
    return v_uint8x16(vnsrl_vx_u8m1(v16, 0, 16));
}

It reuses the register to translate the data from a, b, c, d.

Is there any intrinsic interface to fix this issue now ?
or we need to add more flexible intrinsic to fix this ?

rjiejie, 2021-05-31:
@kito-cheng Please check this issue also, thanks :)

topperc, 2021-06-01:
Is this the same request as #83 ?

topperc, 2021-06-01:
> to fix this memory issue, we designed the some intrinsics "vset_u32m4" like "vector_tuple_types":
>
> inline v_uint8x16 v_pack_b(const v_uint32x4& a, const v_uint32x4& b, [...]

Is the expectation that you will only run this code on a CPU with 128 bit vector registers? Concatenating vector registers and reinterpeting them as a larger vector only works if all the registers are full.

rjiejie, 2021-06-02:
> Is this the same request as #83 ?

not same, I think my interface is more flexible to cover same size or not, in this case, the source is m1 only and dest is m4

topperc, 2021-06-02:
> > Is this the same request as #83 ?
> 
> not same, I think my interface is more flexible to cover same size or not, in this case, the source is m1 only and dest is m4

Sorry I meant is it solving the same underlying problem such that we should have one discussion about the best way to solve it.

rjiejie, 2021-06-02:
> Concatenating

> Is the expectation that you will only run this code on a CPU with 128 bit vector registers?

Uncertain, i could use these interface on some case if I know VLEN is constant.

> Concatenating vector registers and reinterpeting them as a larger vector only works if all the registers are full.

> Is the expectation that you will only run this code on a CPU with 128 bit vector registers?

Not necessarily, it converts 4 m1 vars into 1 m4 vars in this example,
and could merge  any M1 type into Mn without VLEN.

> Concatenating vector registers and reinterpeting them as a larger vector only works if all the registers are full.

These interfaces merge M1 into Mn without computing value like vmov.

rjiejie, 2021-06-02:
> > > Is this the same request as #83 ?
> > 
> > 
> > not same, I think my interface is more flexible to cover same size or not, in this case, the source is m1 only and dest is m4
> 
> Sorry I meant is it solving the same underlying problem such that we should have one discussion about the best way to solve it.

I think it's similar, as #83 mentioned, we can use the interface "vget_*" to change "vfloat32m2x2_t" to "vfloat32m2_t" and
then use the other interface as I mentioned before "vset_f32m4" to merge "vfloat32m2_t" to "vfloat32m4_t"

topperc, 2021-06-02:
> Uncertain, i could use these interface on some case if I know VLEN is constant.
>
>
>
> Not necessarily, it converts 4 m1 vars into 1 m4 vars in this example,
> and could merge any M1 type into Mn without [...]

In your example if VLEN is 256, each m1 register will only have half of the available elements occupied. When they are concatenated together into m4 there will be gaps where the unused elements from the m1 registers were. Using an m4 operation with vl=16 will expect all the elements to be tightly packed into the lower half of the m4 vector.

rjiejie, 2021-06-03:
> In your example if VLEN is 256, each m1 register will only have half of the available elements occupied. When they are concatenated together into m4 there will be gaps where the unused elements from [...]

After deep thinking, you are right, the main task of intrinsic what I mentioned focus whole register pack or extract from one Mx type to another. also useful for the issue in #83, because for it's more flexible :)
it improves performance without memory operation at some constant VLEN condition.

it can not pack data with any VLEN indeed, developer maybe use other instruction like "vcompress" to fix it.

rjiejie, 2021-06-03:
@topperc any comment ?

topperc, 2021-06-03:
#83 was proposing to use the vset/vget/vcreate already defined for tuple and then reinterpret casting between a tuple and regular vector. I think your approach just combines that into one step with new vset/vget?

#95 proposes to remove the tuples which would make #83 not workable. In which case I was already thinking about implementing your proposal to satisfy the underlying desire for #83.

rjiejie, 2021-06-07:
> #83 was proposing to use the vset/vget/vcreate already defined for tuple and then reinterpret casting between a tuple and regular vector. I think your approach just combines that into one step with [...]

Thanks, hope that we will use the intrinsic to fix the issue what we mentioned here and #83 soon :)

rjiejie, 2021-06-09:
@topperc @kito-cheng

Have any progress about for this issue/case ?

rjiejie, 2021-06-10:
@topperc @kito-cheng 

Hello ?

topperc, 2021-06-10:
I think we should implement your proposal. I think we can only do it when the types involved have LMUL>=1. Does that sound right? Extracting a fractional LMUL or inserting a fractional LMUL requires a vslide unless it is to the lowest elements, but we already have lmul extend and truncate intrinsics defined for that case.

If you're willing maybe you could make a pull request for the spec update here? That might get it done sooner than waiting for me or @kito-cheng to do it.

rjiejie, 2021-06-10:
Okay & I will make PR for you

rjiejie, 2021-06-11:
@topperc Please review the PR:
https://github.com/riscv/rvv-intrinsic-doc/pull/99

Thanks.

rjiejie, 2021-06-15:
@topperc Review new commit please :)

topperc, 2021-06-23:
Can this be closed now?

topperc, 2021-06-23:
LLVM patch https://reviews.llvm.org/D104822

rjiejie, 2021-06-24:
> LLVM patch https://reviews.llvm.org/D104822

Nice :)

closed 2021-06-24 by rjiejie