riscv-non-isa/riscv-rvv-intrinsic-doc #95: Proposal for alternate interface for segment load/store and tuple types.

topperc, 2021-06-01:
We've had some implementation issues supporting the tuple types in clang and LLVM IR. This is due to lack of support for structures of scalable vectors in LLVM IR. AArch64 SVE was able to work around this limitation, but the existence of LMUL for RISCV has made it impossible to adopt their approach.

We are able to represent the load/store intrinsics in IR without issue, but the problems show up when we need to store the tuple to a local variable to implement the assignment operator for tuple types. Or when we need to load the tuple from the local variable to use it.

To avoid these implemention issues we at SiFive would like to propose a new interface that avoids the tuples and removes  the `vget`, `vset`, `vcreate`, and `vcopy` intrinsics. We're looking for feedback on this.

For segment stores, we propose to pass between 2 and 8 vectors as separate arguments to the intrinsic rather than a tuple. This combines `vcreate` and store into a single intrinsic. For example

```
void vsseg2e8_v_i8mf8 (int8_t *base, vint8mf8_t v0, vint8mf8_t v1, size_t vl);
void vsseg8e8_v_i8mf8_m (vbool64_t mask, int8_t *base, vint8mf8_t v0, vint8mf8_t v1, vint8mf8_t v2, vint8mf8_t v3, vint8mf8_t v4, vint8mf8_t v5, vint8mf8_t v6, vint8mf8_t v7, size_t vl);
void vssseg2e8_v_i8mf8 (int8_t *base, ptrdiff_t bstride, vint8mf8_t v0, vint8mf8_t v1, size_t vl);
void vssseg8e8_v_i8mf8_m (vbool64_t mask, int8_t *base, ptrdiff_t bstride, vint8mf8_t v0, vint8mf8_t v1, vint8mf8_t v2, vint8mf8_t v3, vint8mf8_t v4, vint8mf8_t v5, vint8mf8_t v6, vint8mf8_t v7, size_t vl);
```

Loads are more complicated because we can't return multiple values in C and we can't make a struct of scalable vectors in C either. We propose to make the intrinsics return void and pass the address of local variables to store the results in 2-8 pointer arguments. These pointers cannot be null. Maskedoff values will be passed as 2-8 additonal arguments. For example:

```
void vlseg2e8_v_i8mf8 (vint8mf8_t *v0, vint8mf8_t *v1, const int8_t *base, size_t vl);
void vlseg8e8_v_i8mf4_m (vint8mf4_t *v0, vint8mf4_t *v1, vint8mf4_t *v2, vint8mf4_t *v3, vint8mf4_t *v4, vint8mf4_t *v5, vint8mf4_t *v6, vint8mf4_t *v7, vbool32_t mask, vint8mf4_t maskedoff0, vint8mf4_t maskedoff1, vint8mf4_t maskedoff2, vint8mf4_t maskedoff3, vint8mf4_t maskedoff4, vint8mf4_t maskedoff5, vint8mf4_t maskedoff6, vint8mf4_t maskedoff7, const int8_t *base, size_t vl);
```

An additional restriction is that tuples can no longer be passed as arguments or return values. I don't think that was likely to be a common scenario.

We would need to add intrinsics independent of tuple types to handle the request from #83.

We believe we can implement this is in clang quite quickly on top of the existing intrinsic support and have it available in LLVM/clang 13 this fall.

We look forward to your feedback.

jan-wassenberg, 2021-06-02:
I understand the difficulties with structs of sizeless vectors, think this proposal makes sense, and have no concerns about usability.

kito-cheng, 2021-06-03:
Although I really don't like the new interface, but I can understand how difficulty for the compiler infrastructure change, so + 0.5 from me.

rdolbeau, 2021-06-03:
First, I'm worried about usability. A load to multiple pointers look really weird and un-user-friendly. It's also relying on the compiler being able to properly allocate registers, and in register-constrained situation that's not a given...

I've used the EPI implementation (from BSC's @rofirrim, see for instance https://github.com/rdolbeau/EPI-test-codes-vector/blob/56a364f9222e83e8a0f2673110ef6f68478ddaf2/simplefft16-rvv/test_fft16_zgemv.c#L160), it's using a tuple interface with no visible issue (I did not investigate how it would mesh with LMUL<>1, I admit).

Another issue is that loosing tuples mean loosing some useful capabilities; for instance, an interesting use case is here: https://github.com/rdolbeau/fftw3/blob/d856efb28255f99212d9553cf15cb36d7eba6a84/simd-support/simd-r5v-split.h#L73-L75, where complex-valued vector are implemented as a tuple. This simplify the code greatly compared to the 'interleaved' implementation (real and imaginary part in alternating lanes of a single vector, of course, at the cost of register pressure...) and can leverage Zvlsseg. Without either structures, tuples or arrays, the format is unlikely to be implementable in FFTW3 whose interface (dated, I know) assume it can return vector of complex values.

For me, removing the tuples is not just a question of a handful of intrinsics changing interface; it's really about being competitive programming-wise by having the ability do do some minimalist 'structuring' with the vectors (at the time, I had asked BSC for structures or fixed-size arrays of vectors; the tuples were the compromise they offered and it was satisfactory).

rofirrim, 2021-06-03:
> I've used the EPI implementation (from BSC's @rofirrim, see for instance [...]

I'd like to give some context here. While we (BSC) were able to provide a convenient interface in the context of EPI, we (SiFive and BSC) deemed the (BSC) implementation too unsafe/unsound/risky for LLVM upstream.

@Hsiangkai  and @topperc (SiFive) have been working hard looking for alternatives, including making proposals LLVM upstream, in order to provide an interface like the `vget`/`vset` one. My understanding is that those alternatives haven't been meet with enthusiasm upstream.

I understand we should not compromise an interface because of compiler limitations but at least this can be implemented in a way that is not too controversial and in a reasonable amount of time. From it we can gather some experience and then determine evaluate again if a better interface is needed (even if this time it means pushing for fundamental changes in LLVM IR itself).

rdolbeau, 2021-06-03:
@rofirrim Well it worked so well for me I didn't realize it was "unsafe/unsound/risky" :-)

Anyway I would ask a question: which intrinsics are using tuples now? Because if it's only Zvlsseg, then my suggestion would be to remove all Zvlsseg-related intrinsics (and the tuples) from the current draft/standard to not delay inclusion of the 'primary' V extension, document that a clean solution is still needed for Zvlsseg (if I'm honest, this looks more like a quick'n'dirty hack to have Zvlsseg than an actual solution...), and document some basic inline assembly examples to facilitate interoperability between the intrinsics and the 'missing' instructions in Zvlsseg - those examples will be needed anyway for people who prefer ASM (... I assume proper constraint for consecutive registers [groups] will exist and allow such inline assembly?)

That way, V intrinsics aren't delayed, and the inclusion of tuples (or structures, or fixed-width array, or ...) can be pushed back to when a clean solution is found/agreed upon by all interested parties.

topperc, 2021-06-03:
Only Zvlsseg are using tuple types today.

I'm not sure there's a clean inline assembly interface here. There's no C variable type you can pass to a single constraint to represent all the segments. Constraints are independent so you can't have multiple constraints and say that one must be the register after another.

@rofirrim did the tuple types in the BSC implementation work with inline assembly?

rofirrim, 2021-06-03:
> @rofirrim Well it worked so well for me I didn't realize it was "unsafe/unsound/risky" :-)

True but your case was reasonably well understood.

> @rofirrim did the tuple types in the BSC implementation work with inline assembly?

No, it didn't.

rdolbeau, 2021-06-03:
@topperc Worst case scenario, you hard-wire your variables to a given register pair but that's not pretty... 

```
__epi_1xf64 tmp0 asm("v4");
__epi_1xf64 tmp1 asm("v5");
asm("vlseg2e64.v %0, (%2)" : "=v" (tmp0), "=v" (tmp1) : "m" (ptr));
```

That's a workaround rather than a (not-good) solution, so there's still hope for a good solution eventually. Not-good solution tends to stick around a lot longer than anticipated :-(

rofirrim, 2021-06-03:
> Only Zvlsseg are using tuple types today.
>
> I'm not sure there's a clean inline assembly interface here. There's no C variable type you can pass to a single constraint to represent all the segments. [...]

Well there could be one, right? I don't think LLVM would have a problem here in the backend.  The register classes for tuples do exist. The issue is more on the LLVM IR side of things where we can't represent the tuple because struct falls short. 

(Maybe in the future we need a `tuple` type in the IR as a first-class type equivalent to `struct` only useable via `extractvalue`/`insertvalue`. No geps or other offset trickery. :smiley:. This is a joke don't take it seriously )

topperc, 2021-06-03:
> Well there could be one, right? I don't think LLVM would have a problem here in the backend. The register classes for tuples do exist. The issue is more on the LLVM IR side of things where we can't [...]

Yes the IR is a problem. Also inline assembly goes through SelectionDAG and uses MVT types to map the constraints to register classes and MVT can't represent tuples either.

rdolbeau, 2021-06-08:
I'm not sure the decision about the interface should be made based on what a specific compiler can or cannot do at a specific point in time. For instance, could e.g. GCC implement the tuples? Could it be done in LLVM with some hypothetical changes in the IR that would be acceptable policy-wise?

A clean but as-yet-unimplemented solution will push the compilers to improve whereas a quick'n'dirty workaround, for all its immediate usefulness, likely won't.

Registers of non-architecturally-specified-width are already in SVE and V, they're likely here to stay. Compilers (and maybe even languages) will have to figure out how to enable them for the users, and that includes the ability to have them in tuples at first, and presumably in structures/array/classes/... later as well.

topperc, 2021-06-08:
I believe gcc was able to implement the tuples. @kito-cheng would know for usre.

We posted patches to llvm to implement tuples earlier this year. They were not met very favorably. We were pointed to conversations AArch64 had a few years ago where longtime llvm developers indicated they were against the idea of scalable vectors in structs which is what is required to support tuples in llvm IR. Those developers aren't as active these days and scalable vectors have been around for a few years and maybe aren't as scary now, but so far we haven't received much support to go against those earlier objections.

AArch64 was able to implement tuples in llvm by using vector types that are 2x, 3x, or 4x as large as their regular vectors rather than structs. This doesn't work for LLVM due to LMUL creating 7 different sizes of vectors so it becomes ambiguos whether a 4x vector type is 4 lmul=1 values or 1 lmul=4 value. I thought I might still be able to make something work with maybe bad register allocation, but I couldn't figure out how to make it work with tuples of fractional lmul.

We acknowledge that the solution proposed is a quick and dirty hack, but we wanted to get something people could use in the near term rather than waiting another year until we maybe had a good solution. I don't believe telling people they need to use inline assembly is a good workaround. Inline assembly can be intimidating and confusing. Especially since you would need to do manual register allocation for these instructions.

rdolbeau, 2021-06-08:
@topperc
>  whether a 4x vector type is 4 lmul=1 values or 1 lmul=4 value

I see that it could be a problem - _if_ the storage is different. I haven't had time to follow in details, but is it different these days? If it's just LMUL consecutive registers vs. N consecutive registers, it may not matter if you can't tell the difference when LMUL == N as they are effectively the same anyway. Didn't look at the problem in details so I could be way off the mark here.

> waiting another year until we maybe had a good solution

Is is really that urgent? V isn't even ratified, I don't think there's commercially available silicon yet, and the handful of IP implementations I'm aware of don't have Zvlsseg anyway... (and usually target pre-v0.10 as it takes a while to implement a core with V support). I know some people already write code for V (I have, as mentioned earlier), but I don't think it's the kind of people that are going to be worried by an ASM wrapper for that one specific extension-to-the-extension if there's a clean solution in the work.

kito-cheng, 2021-06-08:
To GCC, I believe both interface are implement-able, and tuple type are implemented last year, that because AArch64 support struct with scalable vector on GCC IR.

rjiejie, 2021-06-17:
> For segment stores, we propose to pass between 2 and 8 vectors as separate arguments to the intrinsic rather than a tuple. This combines `vcreate` and store into a single intrinsic. For example
>
> ``` [...]

How ugly the interfaces are，hope these are not present in the toolchain, include LLVM

Hsiangkai, 2021-06-21:
> How ugly the interfaces are，hope these are not present in the toolchain, include LLVM

I am not sure if it is more beautiful to use 8 arguments `vcreate` then use `vssseg` to store the tuple. The reality is LLVM IR could not deal with scalable struct in the current status. I ever proposed a RFC[1] to expand the semantic of LLVM IR in the llvm-dev mailing list. No feedback for the proposal. Craig and I joined the ARM SVE sync-up meeting to get feedback from ARM compiler engineers. They involved the development for scalable types in the LLVM earlier than us. The answer we got is they also ever proposed similar idea about 2 years ago and the idea is rejected by the LLVM IR experts at that time. I did not object to the idea of tuple types. Actually, we have a workable downstream version for tuple types. I created several proof of concept patches[2] to demonstrate the idea in the RFC. We are working on an open standard. It is unacceptable we could not upstream the implementation for the open standard in my point of view. We need a feasible solution in the current stage. To remove tuple types is a solution. Do you have any other better suggestions to resolve the issue?

[1] [llvm-dev] [RFC] Permit load/store/alloca for struct containing all scalable vectors. https://lists.llvm.org/pipermail/llvm-dev/2021-March/149296.html
[2] Proof of concept patches:
https://reviews.llvm.org/D97264
https://reviews.llvm.org/D98161
https://reviews.llvm.org/D98169
https://reviews.llvm.org/D99482
https://reviews.llvm.org/D99590
https://reviews.llvm.org/D99592
https://reviews.llvm.org/D99593

ChunyuLiao, 2021-06-25:
I agree with this approach.

gquintin, 2021-09-03:
Hi everyone, I am the [NSIMD](https://github.com/agenium-scale/nsimd) maintainer and we are currently adding support for RISC-V extension "V" into NSIMD.

We propose several APIs (when possible depending on how the headers are written):
- C89 (usually GCC headers are more C89 than Clang ones)
- C99
- C11
- C++98
- C++11
- C++14
- C++20

All the C++ APIs are just templated/operator overloaded wrappers aroung the C APIs. And all our tuple API is based upon structs  from the C APIs up to the C++ APIs:

```c++
template <NSIMD_CONCEPT_VALUE_TYPE T, int N, NSIMD_CONCEPT_SIMD_EXT SimdExt>
NSIMD_STRUCT packx2 {
  typedef typename simd_traits<T, SimdExt>::simd_vector simd_vector;
  typedef T value_type;
  typedef SimdExt simd_ext;
  static const int unroll = N;
  static const int soa_num_packs = 2;

  pack<T, N, SimdExt> v0;
  pack<T, N, SimdExt> v1;
};
```

The `NSIMD_STRUCT` macro is `struct` except when compiling for Arm SVE in which case it is `__sizeless_struct`. So it would break NSIMD's public API to change this paradigm and I must admit that I am really not happy about that.

Another reason that I need to box SIMD/vector types into structs is for operator overloading. Indeed I cannot rely on compilers extensions to provide this kind of features because:

- Some compilers do not provide this kind of overloading
- Sometimes it is just impossible (Intel `__m128i` types can represents any SIMD vector of signed/unsigned integers with different sizes)
- It is non standard C/C++ so I avoid it

Anyway I need to box SIMD/vector types into struct for operator overloading in C++ and function overloading in C++ and C11. Summing up my long post I just need a `__sizeless_stuct` keyword and I will adapt the internals of NSIMD to the API that you decide for tuples (structs or opaque types with getters and setters).

My opinion on this subject concerning upstream LLVM is the same as @rdolbeau. I know it is easy to say but I think you should stick to the design of the API and make as good as possible without taking into account what LLVM IR can or canont do. Arm is pushing really hard SVE so I think that at one point they will be forced to support sizeless types.

eopXD, 2022-08-02:
The interface is now implemented as this issue had proposed. Document is updated in #98. Closing the issue.

closed 2022-08-02 by eopXD