riscv-non-isa/riscv-rvv-intrinsic-doc #96: Intrinsic without side-effect may be deleted by DCE unexpectedly.

pshung, 2021-06-02:
Follow #60 

Giving a test case to demonstrate that intrinsic without side-effect will be removed by DCE.

The code in assembly form produces the desired result.

```c
void ASM(const int32_t* a, int32_t* b, size_t n){
      size_t vl;
      size_t N16 = 16;
      __asm__ __volatile__ ( "vsetvli %[vl], %[N16], e32, m1, tu, mu" : [vl] "=r"(vl) : [N16] "r"(N16));
      int32_t x = INT32_MAX;
      vint32m1_t v;
      __asm__ __volatile__ ( "vmv.v.x %[v], %[x]" : [v] "=vr"(v): [x] "r"(x));
                                                                                                                  __asm__ __volatile__ ( "vsetvli %[vl], %[n], e32, m1, tu, mu" : [vl] "=r"(vl) : [n] "r"(n) );
    __asm__ __volatile__ ( "vle32.v %0, (%1)" : "=vr"(v) : "r"(a) : "memory" );
      __asm__ __volatile__ ( "vsetvli %[vl], %[N16], e32, m1, tu, mu" : [vl] "=r"(vl) : [N16] "r"(N16));    XX    __asm__ __volatile__ ( "vse32.v %[v], (%[b])" : [v] "=vr"(v) : [b] "r"(b) : "memory" );
  }
```

however, the code in intrinsic form produces incorrect result.
```c
  void INT(const int32_t* a, int32_t* b, size_t n){
    size_t vl = vsetvl_e32m1(16);
    vint32m1_t v = vmv_v_x_i32m1(INT32_MAX, vl);

    vl = vsetvl_e32m1(n);
    v = vle32_v_i32m1(a, vl);

    vl = vsetvl_e32m1(16);
    vse32(b, v, vl);
  }
```
We found the the 'vmv' instruction will be deleted by DCE.

topperc, 2021-06-02:
The intrinsics API is currently unable to model tail undisturbed consistently. So that vmv is considered tail agnostic. We need to resolve #27

pshung, 2021-06-02:
DCE only cares whether the instruction has 'use' or side-effect.
It won't care whether the instruction is with 'ta' or 'tu'.

topperc, 2021-06-02:
> DCE only cares whether the instruction has 'use' or side-effect.
> It won't care whether the instruction is with 'ta' or 'tu'.

Right. To model this correctly we need to a tail undisturbed vle intrinsic that takes the tail undisturbed elements as input so that it won't be dead. Without that there is no guarantee the register allocator would even pick the same register for `v` on both instructions even if hadn't been deleted.

closed 2021-06-18 by pshung