riscv-non-isa/riscv-rvv-intrinsic-doc #102: some problem of GCC toolchain

damonyu1989, 2021-07-01:
I use the rvv intrinsic to do some development resently, and I found some problem of the GCC toolchain.

1. The data type of LMUL < 1  is not defined, such as vint8mf2_t, vint8mf4_t, vint8mf8_t, and so on.
2. The type of fp16  is not supported in C library of the  toolchain.

rjiejie, 2021-07-01:
> I use the rvv intrinsic to do some development resently, and I found some problem of the GCC toolchain.
> 
> 1. The data type of LMUL < 1  is not defined, such as vint8mf2_t, vint8mf4_t, vint8mf8_t, and so on.

See https://github.com/riscv/riscv-gcc/issues/279

> 2. The type of fp16  is not supported in C library of the  toolchain.

See https://github.com/riscv/riscv-gcc/issues/280

eopXD, 2022-07-29:
Looks like the problem is answered. Should be ok to close. Thank you @rjiejie.

closed 2022-07-29 by eopXD