riscv-non-isa/riscv-rvv-intrinsic-doc #110: Is it possible to construct "sizeless" structs with register types as members?

gquintin, 2021-09-02:
Hi RISC-V,

I am the maintainer of [NSIMD](https://github.com/agenium-scale/nsimd) and we are currently implementing the RISC-V extension "V" into NSIMD. We are facing a problem with tuples of vector registers. I assume that, as for Arm SVE, vector types are considered "sizeless". As a consequence we cannot use them as members of standard C/C++ structs. Arm SVE, in their original design (and API) provided the `__sizeless_struct` keyword that allows one to write:

```
__sizeless_struct foo {
  vector_type v0;
  vector_type v1;
  // ...
  vector_type vn;
};
```

We tried to find out an equivalent keyword in `riscv_vector.h` but did not succeed. Going through the code we found the getters and setters to work with the opaque types representing the structs that we are trying to build. So we know that what we want to do can be achieved with getters and setters. But as we designed NSIMD the other way it would incur a change in the public API which I am not that happy to make hence my question.

Thank you for your answer.

gquintin, 2021-09-02:
Update: I just realized that it would disable our C++ APIs (C++98 up to C++20) as we use the boxing technique to the overload standard arithmetic operators:

```c++
int main() {
  nsimd::pack<float> va, vb, vc;
  vc = va + vb;
  return 0;
}
```

And we make heavy use of `__sizeless_struct` to achieve the same goal with Arm SVE.

kito-cheng, 2021-09-02:
Due to some technical issue on LLVM... sizeless struct can't be implemented in LLVM world, so we are not intend to include that into RISC-V world, you can read more information at  https://github.com/riscv/rvv-intrinsic-doc/issues/95

eopXD, 2022-08-02:
Sizeless type has always been a battle. The problem already exist when Arm tries to extend sizeless struct on SVE. We may solve them by adding a language extension, but as @kito-cheng have mentioned, we are limited by implementations of LLVM IR and requires consensus from the whole community. So this issue is currently blocked.

Maybe we can revisit this after the v1.0 release?

eopXD, 2022-08-02:
Closing this issue and redirecting to https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/139. The question essentially boils down to how do we enable sizeless struct in the compiler implementation.

closed 2022-08-02 by eopXD