riscv-non-isa/riscv-rvv-intrinsic-doc #111: some type reinterpret_cast

damonyu1989, 2021-09-03:
1. uint/int->mask, in some case,  we need to init a mask by vmv.v.x, and which needs to be reinterpret_casted to vbool type. 
2. f32->u64

eopXD, 2022-07-27:
> uint/int->mask, in some case, we need to init a mask by vmv.v.x, and which needs to be reinterpret_casted to vbool type.

I think you can use vmseq and for this.

> 2. f32 -> u64

I think you can use `vfcvt.xu.f.v` plus `vzext.vf2` to achieve this.

topperc, 2022-07-29:
vmseq wouldn't be a reinterpret_cast. As I under the request, it's to load a vector of say i8 and then being able to treat those bytes as bits of a mask.

I'm not sure what the f32->u64 is asking for. Wouldn't the right instruction be vfwcvt.xu.f.v?

eopXD, 2022-07-30:
> Regarding the 1st question,

I guessed I misunderstood the question and thought the question was to convert a group of vector registers with i8 onto a mask vector register.

If the question is the other way around as you said, to directly reinterpret data in i8 register to a mask register... You may want to understand [v-spec: Mask Register Layout](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#sec-mask-register-layout) and how [boolean type is expressed in rvv intrinsic](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/master/rvv-intrinsic-rfc.md#mask-types).

> Regarding the 2nd question,

Ah yes. I agree with @topperc that `vfwcvt.xu.f.v` is the right instruction to use.