riscv-non-isa/riscv-rvv-intrinsic-doc #112: cannot find vlseg3e8_v_u8m1

figozhang, 2021-09-07:
i wrote some test code for RVV intrinsic, but it looks like some function cannot find in GCC.
i am using riscv-gnu-toolchain repo and "rvv-intrinsic" branch" to build the GCC.

```
riscv64-unknown-elf-gcc -o test  rgb24_bgr24_test_c.o 
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: rgb24_bgr24_test_c.o: in function `rgb24_bgr24_neon_intr':
/home/rlk/rlk/riscv_trainning/vectors/example_rgb24_bgr24_intrinsic/rgb24_bgr24_test.c:30: undefined reference to `vlseg3e8_v_u8m1'
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /home/rlk/rlk/riscv_trainning/vectors/example_rgb24_bgr24_intrinsic/rgb24_bgr24_test.c:36: undefined reference to `vsseg3e8_v_u8m1'
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: rgb24_bgr24_test_c.o: in function `main':
/home/rlk/rlk/riscv_trainning/vectors/example_rgb24_bgr24_intrinsic/rgb24_bgr24_test.c:70: undefined reference to `rgb24_bgr24_intr'
collect2: error: ld returned 1 exit status
make: *** [Makefile:19: test] Error 1
````

rjiejie, 2021-09-18:
see https://github.com/riscv-collab/riscv-gcc/pull/291
Maybe you need to update the latest gcc source

closed 2021-11-22 by Hsiangkai

sunmin89, 2023-07-01:
Hi, I came across the same problem.

Here is my g++ info
```
sunmin@plct:~/rvv/hellorvv/aa$ riscv64-unknown-linux-gnu-g++ -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-linux-gnu-g++
COLLECT_LTO_WRAPPER=/home/sunmin/bins/riscv-gnu-toolchain-bin-rvv/bin/../libexec/gcc/riscv64-unknown-linux-gnu/13.1.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /home/sunmin/aosp-out/riscv-gnu-toolchain/buildv/../gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/home/sunmin/aosp-out/riscv-gnu-toolchain-bin-rvv --with-sysroot=/home/sunmin/aosp-out/riscv-gnu-toolchain-bin-rvv/sysroot --with-pkgversion=gcc035c5d867 --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=../../gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gcv --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-O2    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2    -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 13.1.0 (gcc035c5d867)
```
When I compile the code it complains:

```
v.h: In function 'void rgb24_bgr24_intr(unsigned char*, unsigned char*, long unsigned int)':
v.h:162:9: error: '__riscv_vlseg3e8_v_u8m1x3' was not declared in this scope; did you mean '__riscv_vlse8_v_u8m1_m'?
  162 |         __riscv_vlseg3e8_v_u8m1x3(src, vl);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
      |         __riscv_vlse8_v_u8m1_m
```
Yes this function  is referenced at `auto-generated\api-testing\vlseg3e8.c`

In fact, I can use functions listed within `01_vector_loads_and_stores_functions.md`
But I failed to use data types or functions defined within `02_vector_unit-stride_segment_load_store_instructions_zvlsseg.md`

Should I include the header files from this repo(overloaded-non-policy.h, non-policy.h, etc.) ?

Could someone give me a hand?

Thanks!!

kito-cheng, 2023-07-04:
@sunmin89 GCC 13 has implement 0.11 vector intrinsic except segment load/store, it will included in next GCC release :)

[1] https://gcc.gnu.org/gcc-13/changes.html#riscv

sunmin89, 2023-07-04:
Thanks!

> GCC 13 has implement 0.11 vector intrinsic except segment load/store, it will included in next GCC release :)

I wish my intrinsic code can  comply to v1.0, your mentioned v0.11 seems a lot out of date.  (：

Besides, I’d be appreciated if someone could explain what's the difference between ordinary `load/store` and `segment load/store`.

In order to  solve this  issue, I switch gcc under riscv-gnu-toolchain to master branch and recompiled my toolchain.
This  `__riscv_vlseg3e8_v_u8m1x3`  load interface can compile with success, and it seems working well so far.

While the `__riscv_vsseg3e8_v_u8m1x3` store interface can be compiled too, but it doesn't work as I expected.

dzaima, 2023-07-05:
> I wish my intrinsic code can comply to v1.0, your mentioned v0.11 seems a lot out of date. (：

the v0.11 is the intrinsic API version, not the RVV version number; intrinsics v0.11 does target rvv1.0.

nick-knight, 2023-07-05:
> Besides, I’d be appreciated if someone could explain what's the difference between ordinary load/store and segment load/store.

Here's what the ISA spec says: https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#78-vector-loadstore-segment-instructions . 

You can view the "ordinary" load/store instructions as the special case of segment load/store instructions with NFIELDS = 1. (Indeed, this is how they are encoded.)