riscv-non-isa/riscv-rvv-intrinsic-doc #113: `vint32mf2_t` and some other types seems violate the constraint `LMUL ≥ SEW/ELEN` when `ELEN` = 32

QuarticCat, 2021-09-22:
Is this intentional or did I get something wrong?

This constraint may come from the requirement in the spec that [VLEN >= ELEN](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#2-implementation-defined-constant-parameters).

topperc, 2021-09-22:
You are correct. The compiler will need to disable the intrinsics that use those types when ELEN=32. We can't just remove the types from the spec just because ELEN might be 32. They are needed to support some widening/narrowing conversions for ELEN>=64. So it needs to be determined at compile time if the types exist.

QuarticCat, 2021-09-22:
So the table below needs some modifications?

https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/master/rvv-intrinsic-rfc.md#data-types

topperc, 2021-09-23:
Yes https://github.com/riscv-non-isa/rvv-intrinsic-doc/pull/114

closed 2021-09-24 by QuarticCat