riscv-non-isa/riscv-rvv-intrinsic-doc #115: Redundancy about for "LMUL truncate and extension functions"

rjiejie, 2021-09-28:
I realized these functions is redundancy compared with my PR [1] long times ago about for "Vector Insertion/Extraction Functions".

I think the "Vector Insertion/Extraction Functions" supply minimal operations, and all "LMUL truncate and extension functions"
could wrap insert/extraction functions to implement the same requirements, so they are not necessary,
the spec should only expose the necessary and minimal operation functions for one functional class, is it right ?
so I suggest that we should remove these truncate/extension functions.

[1]: https://github.com/riscv-non-isa/rvv-intrinsic-doc/pull/99

rjiejie, 2021-09-28:
@aswaterman @kito-cheng @topperc 
Any thoughts ?

topperc, 2021-09-28:
The "Vector Insertion/Extraction Functions" don't support inserting or extracting a fractional LMUL value because it doesn't make sense for any index except 0.  Would you propose to add them and require the index be 0?

Using vset for LMUL extension is a bit more verbose since you need to pass vundefined() for the destination.

rjiejie, 2021-09-29:
> The "Vector Insertion/Extraction Functions" don't support inserting or extracting a fractional LMUL value because it doesn't make sense for any index except 0. Would you propose to add them and [...]

Yes, we could use index 0 to emulate, but the fractional LMUL looks like a problem for "Insertion/Extraction" indeed :(

eopXD, 2022-08-02:
Currently GCC and LLVM have implemented them and they may already attract some users to them. We may not have much benefit to prune them now. Maybe we can just leave it there?

Lets reach a decision on this in the v1.0 release?

jan-wassenberg, 2022-08-17:
One input: we do indeed use lmul_trunc and ext in Highway, including for getting vuint8mf4_t as the the lower half of an vuint8mf2_t etc. It would be nice to keep them :)

eopXD, 2023-10-23:
We will have them to stay inside the spec.

closed 2023-10-23 by eopXD