riscv-non-isa/riscv-rvv-intrinsic-doc #117: rename overloaded_intrinsic_funcs.md

HanKuanChen, 2021-09-30:
I notice the overloaded intrinsic function name is very unfriendly to C++ template.
For example, if we want to do element-wise max for two array, we will write
```
#include <riscv_vector.h>

vint32m8_t vle_m8(const int32_t *ptr, size_t vl)
{
    return vle32_v_i32m8(ptr, vl);
}

vint32m8_t max(const int32_t *lhs, const int32_t *rhs, size_t vl)
{
    return vmax(vle_m8(lhs, vl), vle_m8(rhs, vl), vl);
}
```
To make it support unsigned and float, we will write
```
#include <riscv_vector.h>

vint32m8_t vle_m8(const int32_t *ptr, size_t vl)
{
    return vle32_v_i32m8(ptr, vl);
}

vint32m8_t max(const int32_t *lhs, const int32_t *rhs, size_t vl)
{
    return vmax(vle_m8(lhs, vl), vle_m8(rhs, vl), vl);
}

vuint32m8_t vle_m8(const uint32_t *ptr, size_t vl)
{
    return vle32_v_u32m8(ptr, vl);
}

vuint32m8_t max(const uint32_t *lhs, const uint32_t *rhs, size_t vl)
{
    return vmaxu(vle_m8(lhs, vl), vle_m8(rhs, vl), vl);
}

vfloat32m8_t vle_m8(const float *ptr, size_t vl)
{
    return vle32_v_f32m8(ptr, vl);
}

vfloat32m8_t max(const float *lhs, const float *rhs, size_t vl)
{
    return vfmax(vle_m8(lhs, vl), vle_m8(rhs, vl), vl);
}
```
However, they cannot write in a simple C++ template, since max has different name (`vmax`, `vmaxu`, `vfmax`).
```
#include <riscv_vector.h>

vint32m8_t vle_m8(const int32_t *ptr, size_t vl)
{
    return vle32_v_i32m8(ptr, vl);
}

vuint32m8_t vle_m8(const uint32_t *ptr, size_t vl)
{
    return vle32_v_u32m8(ptr, vl);
}

vfloat32m8_t vle_m8(const float *ptr, size_t vl)
{
    return vle32_v_f32m8(ptr, vl);
}

template<class T>
auto max(T lhs, T rhs, size_t vl)
{
    // return vmax(vle_m8(lhs, vl), vle_m8(rhs, vl), vl); // only valid for signed
    // return vmaxu(vle_m8(lhs, vl), vle_m8(rhs, vl), vl); // only valid for unsigned
    // return vfmax(vle_m8(lhs, vl), vle_m8(rhs, vl), vl); // only valid for float
}
```

We should make intrinsics have same name if they have same semantic.
For example,
`vadd`, `vfadd` -> `vadd`
`vwadd`, `vfwadd` -> `vwadd`
`vmax`, `vmaxu`, `vfmax` -> `vmax`
`vmseq`, `vmfeq` -> `vmseq`
`vredsum`, `vfredusum` -> `vredsum`

Old intrinsics will be kept (backward compatibility).

Hsiangkai, 2021-09-30:
I am not sure if we should provide such intrinsics for C++ template. In the previous discussion #7, we decide to provide 1-to-1 intrinsics. Is it possible to provide wrapper functions in users' code to meet the purpose?

HanKuanChen, 2021-09-30:
1. If users want to use 1-to-1 intrinsic, they can use intrinsics which is in intrinsic_funcs.md. What I want to rename is only overloaded intrinsics, which is an optional implementation for compiler.
2. The amount of riscv vector type is huge, this would be a huge effort for users to do this. Meanwhile, if users want to use C++ template, all of them will face the same problem.

Hsiangkai, 2021-09-30:
It may make sense to define them in the overloaded interface. Wait for others' opinions.

kito-cheng, 2021-10-01:
+1 from me.

eopXD, 2022-08-02:
This is a good feature to have. Let's revisit this after the v1.0 release.