riscv-non-isa/riscv-rvv-intrinsic-doc #126: Function prototype of vmerge and vadc-family instructions

eopXD, 2021-11-23:
I found that although the two operations share the same suffix `vvm` / `vxm`, their function prototype are inconsistent unlike `vv` / `vx` operations. Is there a specific reason behind this?

```
vint32m1_t vadc_vvm_i32m1 (vint32m1_t op1, vint32m1_t op2, vbool32_t carryin, size_t vl);
vint16m1_t vmerge_vvm_i16m1 (vbool16_t mask, vint16m1_t op1, vint16m1_t op2, size_t vl);
```

zakk0610, 2022-02-07:
The intrinsic names are as close as the assembly mnemonics. 
`MNEMONIC ::= Instruction name in v-ext specification. Replace '.' with '_'.`
`vadc`/`vmerge` are `vadc.vvm`/`vmerge.vvm`.

eopXD, 2022-02-08:
Thanks for the reply.

closed 2022-02-08 by eopXD

lhtin, 2022-02-22:
> The intrinsic names are as close as the assembly mnemonics. `MNEMONIC ::= Instruction name in v-ext specification. Replace '.' with '_'.` `vadc`/`vmerge` are `vadc.vvm`/`vmerge.vvm`.

I have another question, why vmerge and vader's mask parameter positions different?

nick-knight, 2022-02-22:
I actually have the same question. I think this was the original question, and I don't think it was addressed.

The `v*{ad,sb}c.v*m` instructions have a similar property to the `vmerge.*` instructions: they are encoded as masked, but operate as if they were unmasked, and treat the input mask as another input vector. (This is why the assembly mnemonics use `v0` instead of `v0.t`.) I think it is inconsistent that the `vmerge_*` intrinsics have the `mask` parameter first, while the `v*{ad,sb}c_v*m_*` intrinsics instead have a `carryin`/`borrowin` parameter third. I think it would make more sense if `vmerge` had its `mask` argument third.

zakk0610, 2022-02-24:
@nick-knight @lhtin Thanks, it makes sense to me.
I'm trying to remember something from https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/29.
It seems we make an exception rule for `vmerge` prototype, let it looks like vmaskop ? va : vb which familiar to C. 
I'm ok to update argument order, and we may need to change `vcompress` too.