riscv-non-isa/riscv-rvv-intrinsic-doc #129: Is it lack the `vmacc_vv_<ty>_tu` intrinsics in `intrinsic_funcs.md`

lhtin, 2022-02-06:
Hi,

Is it lack the `vmacc_vv_<ty>_tu` intrinsics in `intrinsic_funcs.md`, but should be supported according to [Tail Policy in the Intrinsics (rvv-intrinsic-rfc.md)](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/master/rvv-intrinsic-rfc.md#tail-policy-in-the-intrinsics) of multily-add intrinsics?

Best regards,
Tin

zakk0610, 2022-02-07:
Yes, you are right. We will fix it later, thanks!

lhtin, 2022-02-07:
@zakk0610 And the vcompress, vslide[down|up], vmv.s.x intrinsics aslo lack the `_tu` intrinsic. All missing includes:

```
# 11.13. Vector Single-Width Integer Multiply-Add Instructions
vmacc
vnmsac
vmadd
vnmsub

# 11.14. Vector Widening Integer Multiply-Add Instructions
vwmaccu
vwmacc
vwmaccsu
vwmaccus

# 13.6. Vector Single-Width Floating-Point Fused Multiply-Add Instructions
vfmacc
vfnmacc
vfmsac
vfmadd
vfmsub
vfnmsub

# 13.7. Vector Widening Floating-Point Fused Multiply-Add Instructions
vfwmacc
vfwnmacc
vfwmsac
vfwnmsac

# 14.1. Vector Single-Width Integer Reduction Instructions
vredsum
vredmaxu
vredmax
vredminu
vredmin
vredand
vredor
vredxor

# 14.2. Vector Widening Integer Reduction Instructions
vwredsumu
vwredsum

# 14.3. Vector Single-Width Floating-Point Reduction Instructions
vfredosum
vfredusum
vfredmax
vfredmin

# 14.4. Vector Widening Floating-Point Reduction Instructions
vfwredosum
vfwredusum

# 16. Vector Permutation Instructions
vmv.s.x
vfmv.s.f
vslideup
vslidedown
vcompress
```

eopXD, 2022-07-29:
https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/47 have updated a new batch of policy intrinsics. I think the issue is resolved now and we can close the issue :)

closed 2022-07-29 by eopXD