riscv-non-isa/riscv-rvv-intrinsic-doc #131: Question about the LLVM emits vsetvl with `tu, mu` when compiling the `vmacc_vv_i8m1` intrinsics

lhtin, 2022-02-07:
Hi all,

I have a question about that the LLVM compiler emits vsetvli with `tu, mu`  when compiling the `vmacc_vv_i8m1` intrinsics (i.e. has dest parameter). Because from the RFC doc, it should emit `ta, mu`. And there should be a corresponding `_tu` intrinsic which should emit `tu, mu`. And from my experience about using intrinsic, it can reduce the number of the vsetvli instruction.

[Here](https://godbolt.org/z/c4zoKvcje) can recurrent.

C code:
```c
#include <riscv_vector.h>

vint8m1_t vmacc_test(vint8m1_t dest, vint8m1_t op1, vint8m1_t op2, size_t vl)
{
    vint8m1_t t = vadd_vv_i8m1(op1, op2, vl);
    return vmacc_vv_i8m1(dest, t, op2, vl);
}
```

RVV Assembly:
```
vmacc_test:
        vsetvli zero, a0, e8, m1, ta, mu
        vadd.vv v9, v9, v10
        vsetvli zero, zero, e8, m1, tu, mu
        vmacc.vv        v8, v9, v10
        ret
```

Best regards,
Tin

zakk0610, 2022-02-08:
upstream LLVM had [reverted ](https://reviews.llvm.org/rG7ccd31c900edecb1ec3eb75bb6f917dbe674bc72)"_tu" and "_mt" intrinsics supported due to adding those new set of intrinsics increase clang binary size/clang compilation time. 
https://reviews.llvm.org/D110684#3060191
so before LLVM support that, the default behavior for vamcc are still always tail undisturbed.

BTW, we want to propose this new design https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/120 for tail/mask policy and it's still under reviewing and developing on phabricator. (Ex.  https://reviews.llvm.org/D111617, [D117681](https://reviews.llvm.org/D117681)  and /[D117989](https://reviews.llvm.org/D117989)).

fabianpedd, 2022-07-04:
Are there any updates on the LLVM support for [Policy Intrinsic Functions](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/master/rvv-intrinsic-rfc.md#policy-intrinsic-functions)? I just tried to compile some code with both [GCC](https://github.com/riscv-collab/riscv-gcc/tree/riscv-gcc-rvv-next) and LLVM 14. GCC had no issues with `_tu`, but LLVM 14 dislikes them.

eopXD, 2022-08-02:
Current unmasked `vmacc` is `tu`. So I guess your issue is resolved?

eopXD, 2022-08-06:
@fabianpedd Patches for policy intrinsics have been landed into LLVM. You may compile them now with the trunk LLVM.

closed 2022-08-11 by lhtin