riscv-non-isa/riscv-rvv-intrinsic-doc #134: Is `vl` argument

cooldome, 2022-02-18:
Hi,
I am new to this forum, likely the question has been discussed before, plus point to the relevant thread if it is the case.

I am not clear why many intrinsic functions accept extra `vl` argument and internally containing extra call to `vsetvl` leading to inferior generated code. Assembly instructions do not have `vl` argument and not clear why intrinsics deviate from this approach.

Main problem with extra third argument is not only inferior performance, but also impossibility to write generic code using operator overloading. Can't write `a + b` anymore with operator overloading, you now need to pass third argument somehow. So it either have to be `plus(a, b, vl)` which is very poor, or you have to pack `vl` into every number. Hence `a` becomes a struct with `vl` and `b` also contains `vl` in its struct and data structures become bloated.

nick-knight, 2022-02-18:
Here's a relevant discussion from a while back: https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/8 . 

An early RVV "intrinsics" (a C language dialect, really), predating this repo, was developed internally at SiFive in mid-2019 and prototyped on a GCC branch later that year. That version followed the "implicit VL" approach you're suggesting. Some of us, like myself, preferred implicit-VL because it more closely matches the underlying assembly language and the API is (arguably) simpler. However, the compiler folks argued that compilers struggle so greatly with global state that performance would ultimately be best using explicit-VL.

The truth is, when you commit to using the C language, you are relinquishing some degree of control to your compiler. It's a double-edged sword. I argue that if you expect your compiler to schedule the (generated) vector assembly to achieve good performance on an arbitrary target, you'll need to grant your compiler the flexibility to change `vl` and `vtype` beyond the explicit `vsetvl`s in your C+intrinsics source code. (It took me over a year to come to accept this, I'm not claiming it's an easy pill to swallow.)

Regarding "inferior code", frankly I've found LLVM's `vsetvl`-elision pass to be rather powerful. Sure, I've found a few minor blemishes, like the tail-undisturbed FMA issue, but I have taken the time to report them, and I can confirm they being fixed. If you have a concrete example of inferior code, please report it, not just for your benefit but for the good of our whole community!

I agree with you that explicit-VL is problematic when trying to build abstractions, like C++ operator overloading, on top. And unfortunately your proposed struct-based workaround won't work because "LLVM doesn't allow sizeless types inside structs". (I don't think GCC has this issue.) I don't have a solution for you, and I doubt there is one within the current constraints.

Another issue I have with explicit-VL, which you didn't mention, concerns how the API will evolve to handle mask and tail behavior (`vta` and `vma` fields of `vtype` CSR), floating-point rounding mode (`frm` CSR), and fixed-point rounding mode (`vxrm` CSR). In each case, the proposed solution has been an "explicit" approach, i.e., adding new intrinsics or new arguments to existing intrinsics. The API is already enormous, and each of these enhancements will only bloat it further.

My attitude is that this API is firmly set on its path forward following the "explicit" approach, especially given the massive compiler engineering undertaking that has occurred over the past 2+ years to support it. I expect someone someday will introduce an alternative "implicit" approach --- it's RISC-V, after all, there's nothing preventing this --- but if they want it to be competitive they better have a small army of compiler engineers lined up to help.

cooldome, 2022-02-19:
Thank you both!
Bery useful summary. I now understand problem much better. Extra `vsetvl` call inside intrinsics not only sets length but also type, so it matches the type of the arguments. Hence, implicit `vl` will also mean implicit `type`, intrinsics will have to be typeless. Implicit state has its disadvantages as well if one can't tell what `a + b` actually does if you know `a` and `b` due to state changes elsewhere is a problem of its own.

closed 2022-07-24 by kito-cheng