riscv-non-isa/riscv-rvv-intrinsic-doc #139: Make the tuple type API for segment load store as compiler optional feature

zakk0610, 2022-03-08:
https://github.com/riscv-non-isa/rvv-intrinsic-doc/pull/137#issuecomment-1061363901

nick-knight, 2022-03-08:
We will probably need to revert https://github.com/riscv-non-isa/rvv-intrinsic-doc/pull/98 too, tagging these functions appropriately.

lhtin, 2022-03-08:
Maybe it's necessary to change the marco name `__riscv_rvv_tuple_type` to `__riscv_v_tuple_type`, because it is unified with [`__riscv_v_intrinsic_overloading`](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/master/rvv-intrinsic-rfc.md#overloaded-interface).

kito-cheng, 2022-03-10:
@lhtin Yeah, `__riscv_v_tuple_type` seems like better naming, thank for your suggestion !

eopXD, 2022-08-02:
Multiple issues have raised request for tuple types and size-less struct and it all boils down if the compiler is able to implement them...

Redirecting other existing issues(#13, #17 , #110 , #154) here for this issue to be the placeholder to discuss on the problem after the v1.0 release.

PR #198 (2023-06-12): Improve segment load/store interface with tuple types
This resolves #139. This change is motivated by constant unpleasant user feedback of the current segment load/store intrinsics and hope to align the interface of segment load/store intrinsics with the non-segment ones.

This PR adds the tuple-type segment load-store intrinsics and removes the existing ones. To company the tuple-types, additional `vget` and `vset` intrinsics for them are added.

Take `vlseg2e8` for example, these are the intrinsics added:

```cpp
vint8mf8x2_t __riscv_vlseg2e8_v_i8mf8x2(const int8_t *base, size_t vl);
vint8mf [... cut]
closed 2023-06-12 by eopXD