riscv-non-isa/riscv-rvv-intrinsic-doc #140: Move mask argument positions to first for merge intrinsic function.

zakk0610, 2022-03-08:
@nick-knight 's [perspective ](https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/126#issuecomment-1048048817) convince me, but if we make the changed, how do users know that when they should use the new one API or old one.

It seems like a general question like, if we decide to change the interface which would break the backward compatible, how to let users to known about that.

eopXD, 2022-03-12:
I think formalizing rvv intrinsic releases will solve the problem.

We can have the new ones and old ones co-exist with deprecation warnings by the compiler. Then remove them in the next release.

eopXD, 2022-08-02:
Since we may already have users on the current intrinsics and don't want to break backward compatibility, maybe we would prefer to extend more intrinsics as your request specifies than replacing them?

Let's hold this for discussion after the v1.0 release?

closed 2023-01-13 by eopXD