riscv-non-isa/riscv-rvv-intrinsic-doc #142: vfredusum_vs_f32m1_f32m1 not supported by gcc10.1

jrahmeh, 2022-03-11:
Is vfredusum_vs_f32m1_f32m1 supported in gcc?  Which version?

zhongjuzhe, 2022-03-12:
I would suggest you to use this version GCC: https://github.com/riscv-collab/riscv-gcc/pull/334. This version support all C api intrinsics in this https://github.com/riscv-non-isa/rvv-intrinsic-doc/tree/new-policy-proposal   proposal

eopXD, 2022-07-29:
We have regular intrinsics both available in latest GCC and LLVM.

```cpp
vfloat32m1_t vfredusum_vs_f32m1_f32m1 (vfloat32m1_t dest, vfloat32m1_t vector, vfloat32m1_t scalar, size_t vl);
vfloat32m1_t vfredusum_vs_f32m1_f32m1_m (vbool32_t mask, vfloat32m1_t dest, vfloat32m1_t vector, vfloat32m1_t scalar, size_t vl);
```

Also policy intrinsics if you care about tail elements of the destination register. These are now available in latest GCC and going to land soon in LLVM.

```cpp
vfloat32m1_t vfredusum_vs_f32m1_f32m1_tu (vfloat32m1_t merge, vfloat32m1_t vector, vfloat32m1_t scalar, size_t vl);
vfloat32m1_t vfredusum_vs_f32m1_f32m1_ta (vfloat32m1_t vector, vfloat32m1_t scalar, size_t vl);
vfloat32m1_t vfredusum_vs_f32m1_f32m1_tum (vbool32_t mask, vfloat32m1_t merge, vfloat32m1_t vector, vfloat32m1_t scalar, size_t vl);
vfloat32m1_t vfredusum_vs_f32m1_f32m1_tam (vbool32_t mask, vfloat32m1_t vector, vfloat32m1_t scalar, size_t vl);
```

closed 2022-07-29 by eopXD