riscv-non-isa/riscv-rvv-intrinsic-doc #144: [RFC] Propose rename current fixed-point intrinsic APIs for instructions which read VXRM

arcbbb, 2022-03-18:
As the discussion in https://github.com/riscv/riscv-v-spec/issues/739,
I would like to add 4 suffixes `_rnu`, `_rdn`, `_rne` and `_rod` to current intrinsic name,
For example of `vaadd`, this would increase the number of intrinsics by 4 times.
```diff
- vint8mf8_t vaadd_vv_i8mf8 (vint8mf8_t op1, vint8mf8_t op2, size_t vl);
+ vint8mf8_t vaadd_rnu_vv_i8mf8 (vint8mf8_t op1, vint8mf8_t op2, size_t vl);
+ vint8mf8_t vaadd_rdn_vv_i8mf8 (vint8mf8_t op1, vint8mf8_t op2, size_t vl);
+ vint8mf8_t vaadd_rne_vv_i8mf8 (vint8mf8_t op1, vint8mf8_t op2, size_t vl);
+ vint8mf8_t vaadd_rod_vv_i8mf8 (vint8mf8_t op1, vint8mf8_t op2, size_t vl);
```
Since there is no default value for VXRM on the function entry, I think the name with suffixes can make it explicitly and users will be aware of what is going to be used.
The candidates of this change are VAADD, VAADDU, VASUB, VASUBU, VSMUL, VSSRL, VSSRA, VNCLIP, and VNCLIPU.

nick-knight, 2022-03-18:
Will we follow a similar pattern for `frm` for floating-point intrinsics?

kito-cheng, 2022-03-20:
I would strongly prefer keep current API *and* adding static rounding mode API instead of removing old one to make sure the compatible, and consistent design with floating point operations

arcbbb, 2022-03-21:
> Will we follow a similar pattern for `frm` for floating-point intrinsics?

I think it is yes, and I would like to create a separate issue for the discussion.

> I would strongly prefer keep current API _and_ adding static rounding mode API instead of removing old one to make sure the compatible, and consistent design with floating point operations

Since we have two options:

1. without suffix indicating the round mode, 
2. with suffix indicating the round mode.

Assuming only option (2) is available, I expect the following C code
```C
vint8m1_t test1(vint8m1_t dest, vint8m1_t op1, vint8m1_t op2, size_t vl) 
{
    vint8m1_t t = vaadd_rdn_vv_i8m1(op1, op2, vl);
    return t;
}
```
would generate the assembly like
```asm
test1:
    vsetvli zero, a0, e8, m1, ta, mu
    csrwi vxrm, 2
    vaadd.vv v8, v9, v10
```
Because Option (2) changes the content of VXRM, I think extra instructions would be needed to save/restore to VXRM when option (1) is also available.
For example,
```C
vint8m1_t test2(vint8m1_t dest, vint8m1_t op1, vint8m1_t op2, size_t vl, int cond)
{
    __asm__ __volatile__("csrwi vxrm, 0");
    vint8m1_t t1 = vaadd_rdn_vv_i8m1(op1, op2, vl);
    vint8m1_t t2 = vaadd_vv_i8m1(t1, op2, vl);
    return t2; 
}
```
I expect the assembly would be 
```asm
test2:
        vsetvli zero, a0, e8, m1, ta, mu
        csrwr vxrm, 0
        csrrwr a0, vxrm, 2
        vaadd.vv v8, v9, v10
        csrw vxrm, a0
        vaadd.vv v8, v8,  v10
        ret
```
And users may lose optimization chances, like in this case
```C
vint8m1_t test3(vint8m1_t dest, vint8m1_t op1, vint8m1_t op2, size_t vl)
{
    vint8m1_t t1 = vaadd_vv_i8m1(op1, op2, vl);
    vint8m1_t t2 = vaadd_vv_i8m1(op1, op2, vl);
    return vadd_vv_i8m1(t1, t2, vl);
}
```
vaadd is not CSEed because of the hidden state.
```asm
test3:
    vsetvli zero, a0, e8, m1, ta, mu
    vaadd.vv    v8, v9, v10
    vaadd.vv    v9, v9, v10
    vadd.vv v8, v8, v9
    ret
```
I see ARM drops some intrinsic support when their compiler goes to version 6 in the document
https://developer.arm.com/documentation/dui0774/g/Other-Compiler-specific-Features/ACLE-support
so I hope it is still not too late to make this change if we agree option (2) fits our needs.

zakk0610, 2022-03-21:
> vaadd is not CSEed because of the hidden state.

I think it's user's responsibility, if they care about rounding they need to use all rounding series intrinsics in the same kernel.

> I would like to add 4 suffixes _rnu, _rdn, _rne and _rod to current intrinsic name,
For example of vaadd, this would increase the number of intrinsics by 4 times.

IPO, I like `vaadd_vv_rnu_i8mf8` rather than `vaadd_rnu_vv_i8mf8` because assembly mnemonics is `vaadd.vv`.

> I would strongly prefer keep current API and adding static rounding mode API instead of removing old one to make sure the compatible, and consistent design with floating point operations

 I think compatible is not a problem if we have predefined marco for intrinsic-doc version...
But I like to keep the current API because if users don't care about rounding or they are doing a fast version kernel, what can they do without the current API?

kito-cheng, 2022-04-03:
> vaadd is not CSEed because of the hidden state.

Those fixed pointer operation only read hidden state/vxrm, so that's should not block CSE in theory, that different situation than `vl`/`vtype`, we read AND write that on RVV intrinsic semantic, so they cause CSE optimization issue.

> But I like to keep the current API because if users don't care about rounding or they are doing a fast version kernel, what can they do without the current API?

Keep current API can let dynamic rounding mode work as expect if we don't break that, and not forcing user to always using static rounding mode scheme.

I would like to keep the possibility and let user have freedom to choice what they like if possible.

eopXD, 2022-08-02:
Fixed point intrinsic is a good feature beneficial to the completeness of RVV intrinsic.

Let's open for discussion whether we should include this in v1.0 release.

closed 2023-10-24 by eopXD