riscv-non-isa/riscv-rvv-intrinsic-doc #145: How to support backward incompatibilities API.

zakk0610, 2022-03-22:
We have several proposals which are trying to change the current API (ex. renaming or reorder the argument order), and some of them make the design more consistent and reasonable.

For examples:
1. https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/140
2. https://github.com/riscv-non-isa/rvv-intrinsic-doc/pull/137#issuecomment-1054464593 masked segment loads don't need additional maskedoff parameters
3. Even we could have backward  incompatibility policy API design.

If we agree we want to support the incompatibility API, then we need to have consensus on how to support that.

I have two proposals as shown below:
1. Compiler supports pre-defined macro for RVV API version. 
Compiler could optionally supports different version RVV API  based on rvv-intrinsic-doc tag. For now we have v0.10 and v0.9 tag.  The predefined macro `__riscv_v_intrinsic_version` have supported version information which follows the formula we ready have in https://github.com/riscv-non-isa/riscv-c-api-doc/blob/master/riscv-c-api.md#architecture-extension-test-macro
`<MAJOR_VERSION> * 1,000,000 + <MINOR_VERSION> * 1,000 + <REVISION_VERSION>`
In the future, when we decide to freeze the API then the tag will not be changed any more. But in fact, we still have opportunity to update API because we have version inforation.

```
// user's code
#if __riscv_v_intrinsic_version == 10000
// v0.10 api
#elif __riscv_v_intrinsic_version == 9000
// v0.9 api
#else
#error unsupported rvv intrinsic functions version.
#endif
```
or
```
// user's code
#define __riscv_v_intrinsic_version 9000
#include <riscv_vector.h>

// v0.9 api

````
2. The new set of intrinsic functions naming would follow the rule in https://github.com/riscv-non-isa/riscv-c-api-doc/pull/25.
In addition, new set of intrinsic functions are only supported in the different intrinsic header which follow the rule in https://github.com/riscv-non-isa/riscv-c-api-doc/issues/17.
The new naming will look like `vint8m1_t __rvv_vadd_vv_i8m1(vint8m1_t vs2, vint8m1_t vs1, size_t vl);` and header file name may be `riscv_v_intrinsic.h`.
In this way, we also could do anything we want in the new intrinsic API set. But this is our latest time to update naming(maybe overloading version only), argument number and order, so we need to have comprehensive consideration and review all proposals in the issue list.

Both proposals may require a schedule for API freeze day, and we also need to list what are missing now..

zhongjuzhe, 2022-03-23:
The RVV GCC is in the RISC-V foundation repo but not push to GCC upstream yet. What I have supported in RVV GCC are all latest C api intrinsics that you have described in the documents. So I didn't support the old obsolete intrinsics. Do you want me to support the old obsolete intrinsics before pushing them upstream? I am Ok with that. It I need to do so, Can you list the api both 0.9 version and 1.0 version together in the document?  
For example:
`version 0.9:`
`vmerge_vv_i8m1 (vint8m1_t v0, vint8m1_t v1, vbool8_t mask, size_t vl)`
`version 1.0:`
`vmerge_vv_i8m1 (vbool8_t mask, vint8m1_t v0, vint8m1_t v1, size_t vl)`

If the api are the same both 0.9 and 1.0 version. Just keep the document no need to change.
Besides, I am waiting for the RVV intrinsic document to be frozen.

eopXD, 2022-08-02:
I think we may not want to break backward compatibility in any sense. We may improve the user experience by extending the intrinsic API.

Maybe we can discuss on how extensions can be added to the intrinsic after this v1.0 release?

eopXD, 2022-09-08:
I want to raise the discussion regarding backward compatibility back again. I originally thought that maybe we should defer the problem of fixing these glitches to discussion after v1.0, however after a second thought, maybe v1.0 is the best timing to apply incompatible changes since we have not reached a "formal" release yet (nothing is ratified).

I think that changes that is backed with functional or compiler optimization reason justifies themself.

On the other hand, people may find some glitches in the current intrinsic API-s. For example, issue https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/159 pointed out that there is inconsistency on some functions where it can be expressed in reduce form without ambiguous naming but has output type appended in the function name. Another example is https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/140, where the argument orderings are inconsistent.

Taking this further, we may always come back to the same dilemma, where we consider software stability on one side and improving the API on the other. So I think it is important we should take the courage to talk about this and rule out some SOP that we can follow.

Assuming that an incompatible change have gathered approval from the community, how do we apply it?

A straightforward approach is that the approved incompatible change be applied in the next release. A more conservative approach is we find a way to have deprecation warnings of the change in the next version, and have the change be applied in the next-next version.

I personally think we should take a more conservative approach on this if we decide to formalize an approach to resolve modifications of incompatible change. Please share your thoughts on this.

rjiejie, 2022-10-27:
This issue like Vector ISA, maybe for all ISA version issue,
we need to considerate multi-versions incompatibility.

+1 for separated specific version intrinsic header like as riscv_v_intrinsic_<version>.h
e.g. riscv_v_intrinsic_v0p7.h

closed 2023-10-24 by eopXD