riscv-non-isa/riscv-rvv-intrinsic-doc #148: Vector Add Intrinsic Code

Mousavikia, 2022-04-13:
Hi I have been working on vector intrinsic coding recently and I have written a simple vector add code like this:
```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>

void* memcpy(void* dest, const void* src, size_t n)
{
	for (size_t i = 0; i < n; i++)
	{
		((char*)dest)[i] = ((char*)src)[i];
	}
}

int compare(int* ref, int* actual, int n) {
	int r;
	for (int i = 0; i < n; ++i) {
		if (ref[i] - actual[i] == 0) {
			r = 1;
		}
		else {
			r = 0;
			break;
		}
	}
	return r;
}

// index arithmetic
void add(int* a, int* b, int* c, int n) {
	for (int i = 0; i < n; ++i) {
		a[i] = b[i] + c[i];
	}
}

void vec_add(int* a, int* b, int* c, int n) {
	while (n > 0) {
		size_t vl = vsetvl_e32m1(n);
		vint32m1_t vb = vle32_v_i32m1(b, vl);
		vint32m1_t vc = vle32_v_i32m1(c, vl);
		vint32m1_t va = vadd_vv_i32m1(vb, vc, vl);
		vse32_v_i32m1(a, va, vl);
		a += vl;
		b += vl;
		c += vl;
		n -= vl;
	}

}

int main(void) {
	// data
	int a[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
	int b[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };

	// compute
	int result[31];
	int vecresult[31];
	add(result, a, b, 31);
	vec_add(vecresult, a, b, 31);

	// compare
	uart_puts(compare(vecresult, result, 31) ? "pass" : "fail");
	return 0;
}
```
When I compile it using gcc it gives following warnings:
```
riscv32-unknown-elf-gcc -march=rv32imv -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -Wall   -I/home/kiian/Desktop/VicunaRepos/vicuna/sw//lib/ -c -o vadd.o vadd.c
vadd.c: In function 'vec_add':
vadd.c:37:33: warning: passing argument 1 of 'vle32_v_i32m1' from incompatible pointer type []8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wincompatible-pointer-types-Wincompatible-pointer-types]8;;]
   37 |   vint32m1_t vb = vle32_v_i32m1(b, vl);
      |                                 ^
      |                                 |
      |                                 int *
In file included from vadd.c:2:
/opt/riscv/lib/gcc/riscv32-unknown-elf/10.1.0/include/riscv_vector.h:509:1: note: expected 'const int32_t *' {aka 'const long int *'} but argument is of type 'int *'
  509 | _RVV_INT_ITERATOR (_RVVINTLD)
      | ^~~~~~~~~~~~~~~~~
vadd.c:38:33: warning: passing argument 1 of 'vle32_v_i32m1' from incompatible pointer type []8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wincompatible-pointer-types-Wincompatible-pointer-types]8;;]
   38 |   vint32m1_t vc = vle32_v_i32m1(c, vl);
      |                                 ^
      |                                 |
      |                                 int *
In file included from vadd.c:2:
/opt/riscv/lib/gcc/riscv32-unknown-elf/10.1.0/include/riscv_vector.h:509:1: note: expected 'const int32_t *' {aka 'const long int *'} but argument is of type 'int *'
  509 | _RVV_INT_ITERATOR (_RVVINTLD)
      | ^~~~~~~~~~~~~~~~~
vadd.c:40:17: warning: passing argument 1 of 'vse32_v_i32m1' from incompatible pointer type []8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wincompatible-pointer-types-Wincompatible-pointer-types]8;;]
   40 |   vse32_v_i32m1(a, va, vl);
      |                 ^
      |                 |
      |                 int *
In file included from vadd.c:2:
/opt/riscv/lib/gcc/riscv32-unknown-elf/10.1.0/include/riscv_vector.h:510:1: note: expected 'int32_t *' {aka 'long int *'} but argument is of type 'int *'
  510 | _RVV_INT_ITERATOR (_RVVINTST)
      | ^~~~~~~~~~~~~~~~~
vadd.c: In function 'memcpy':
vadd.c:11:1: warning: control reaches end of non-void function []8;;https://gcc.gnu.org/onlinedocs/gfortran/Error-and-Warning-Options.html#index-Wreturn-type-Wreturn-type]8;;]
   11 | }
      | ^
riscv32-unknown-elf-gcc -march=rv32imv -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -Wall  -T /home/kiian/Desktop/VicunaRepos/vicuna/sw//link.ld  /home/kiian/Desktop/VicunaRepos/vicuna/sw//crt0.o vadd.o /home/kiian/Desktop/VicunaRepos/vicuna/sw//lib/uart.o -o vadd.elf
riscv32-unknown-elf-objcopy -O binary vadd.elf vadd.bin
srec_cat vadd.bin -binary -offset 0x0000 -byte-swap 4 -o vadd.vmem -vmem
rm vadd.bin
```
I know warnings can happen and they don't mean that a code will fail but when I execute this code on a vector core it doesn't give any output. I know this question is more relevant to ask in the repo of that core but I just want to make sure my code is correct or not

eopXD, 2022-04-13:
I can compile your code with Clang and execute successfully.

Replacing `uart_puts(compare(vecresult, result, 31) ? "pass" : "fail");` with `printf("pass: %d\n", compare(vecresult, result, 31));`.

Mousavikia, 2022-04-13:
Thank you for your help... I have clang +llvm as a compiler option as well... so  can you please write the instructions so that I can run and execute the code myself and don't bother you again?
Till now I have used ready makefiles to produce .vmem file and put them on fpga ram... So I don't know much about this...

eopXD, 2022-05-28:
You may compile the code with LLVM trunk, with CMake configured to build target RISCV.
You may execute the code with `qemu-riscv64` built on QEMU.

closed 2022-07-29 by eopXD