riscv-non-isa/riscv-rvv-intrinsic-doc #149: Question about narrowing right shift arithmatic

Mousavikia, 2022-04-15:
Hi I have question about narrowing shift right:
I have written the following code and when I implement it on a vector core uart returns pass so everything looks fine and ok. What I want to do is this: shift right vector element which are 16 bit data by a amount (lets say 1) and put the result inside a 8 bit data type. In vector part first vsetvl is for 16 bit elements but after I implemented the code I thought I should have added another vsetvl before store since now elements are 8 bit... First of all is this narrowing shift really what I have understood or is it different? Second, does compiler takes care of the second vsetvl that I have forgot or something else is happening?
```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>

void* memcpy(void* dest, const void* src, size_t n)
{
	for (size_t i = 0; i < n; i++)
	{
		((char*)dest)[i] = ((char*)src)[i];
	}
}

int compare(signed char* ref, signed char* actual, signed short int n) {
	signed short int r;
	for (signed short int i = 0; i < n; ++i) {
		if (ref[i] - actual[i] == 0) {
			r = 1;
		}
		else {
			r = 0;
			break;
		}
	}
	return r;
}

void shift(signed char* b, signed short int* a, signed short int n, unsigned char amount) {
	for (int i = 0; i < n; ++i) {
		b[i] = a[i] >> amount;
	}
}

void vec_shift(signed char* shift, signed short int* a, signed short int n, unsigned char amount) {
	while (n > 0) {
		size_t vl = vsetvl_e16m1(n);
		vint16m2_t va = vle16_v_i16m2(a, vl);
		vint8m1_t vshift = vnsra_wx_i8m1(va, amount, vl);
		vse8_v_i8m1(shift, vshift, vl);
		shift += vl;
		a += vl;
		n -= vl;
	}

}

int main(void) {
	// data
	signed short int a[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
	// compute
	signed char result[31];
	signed char vecresult[31];
	shift(result, a, 31, 2);
	vec_shift(vecresult, a, 31, 2);
	// compare
	uart_puts(compare(vecresult, result, 31) ? "pass" : "fail");
	return 0;
}
```

eopXD, 2022-04-15:
First of all you should know that `vsetvl` is used as vector length getter and it helps the compiler to generate correct `vsetvl` instruction. For details of the `vsetvl` instruction, you should refer to the v-spec.

As you seem to have way to simulate the program, I would recommend you to use `objdump` and verify if the compiler is inserting correct `vsetvl` to adjust to the correct SEW as expected. If not, please submit a bug report to the compiler.

Mousavikia, 2022-04-17:
Thank you @eopXD I checked the `objdump` and yes it inserts correct `vsetvl `

closed 2022-04-20 by Mousavikia