riscv-non-isa/riscv-rvv-intrinsic-doc #151: Question about C++ overloaded functions of segment load

zhongjuzhe, 2022-04-20:
Here are 2 functions in the C++ overloaded functions:
1.`void vlseg2e8_v_i8mf8 (vint8mf8_t *v0, vint8mf8_t *v1, const int8_t *base, size_t vl);`
2.`void vlseg2e8_tu (vint8mf8_t *v0, vint8mf8_t *v1, const int8_t *base, size_t vl);`

These 2 functions have the same arguments. Why does one can be overloaded but the other can not be?

zakk0610, 2022-04-21:
In fact, `_tu` need to have merge operand just like masked segment operation have maskedoff to align current design.
I will update #137 later to fix all issues in one time.
 
current known issues in #137:

1. TA and TAMA vslidedown/vslideup need the dest operand for correctness. https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/32
2. segment load and vfclass TU policy need to have merge operand.
3. vmv.s.x and vfmv.s.f ta policy have no overloaded api.
4. reduction TAM have no merge operand.

Let me know if anything is missing. I'm really appreciate your time to help and review.

zhongjuzhe, 2022-04-21:
You have remove `vqmacc` in C intrinsic document but overloaded functions document still have `vqmacc`. I think you should remove them. 
By the way, does the RISC-V community agrees https://github.com/riscv-non-isa/rvv-intrinsic-doc/pull/137 this proposal and merge them into master in the future? I understand the explosion of RVV intrinsics will make LLVM compilation time bad due to
the big <riscv_vector.h>. But this is not a problem in GCC. I hope this proposal can be merged into master branch finally.

kito-cheng, 2022-04-21:
@zhongjuzhe https://reviews.llvm.org/D111617 I am working on improve compilation speed for clang/LLVM, I believe that's could resolve that issue.

zakk0610, 2022-04-22:
> I have 2 more questions: 1.You have remove `vqmacc` in C intrinsic document but overloaded functions document still have `vqmacc`. I think you should also remove them in overloaded functions. 2.tamu/tumu/tuma policy in segment load, you don't have `merge` or `maskoff` operand, do you need to add them too?
> 

Thanks, I will remove vqmacc in document later.

for 2, I saw that have `merge` operand for segment load, maybe you checkout the wrong branch.
https://raw.githubusercontent.com/riscv-non-isa/rvv-intrinsic-doc/new-policy-proposal/policy_funcs/intrinsic_funcs/03_vector_load_store_segment_instructions_zvlsseg.md

> By the way, does the RISC-V community agrees #137 this proposal and merge them into master in the future? I understand the explosion of RVV intrinsics will make LLVM compilation time bad due to the [...]

closed 2022-04-22 by zhongjuzhe