riscv-non-isa/riscv-rvv-intrinsic-doc #156: [RFC] Use case for type casting of boolean vector values

eopXD, 2022-08-02:
Issue in #69 points out that current vector mask registers all follow MLEN = 1, which bit values are compact inside a single vector register ([4.5. Mask Register Layout](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#sec-mask-register-layout)). Current bool types exist because of type safe reasons. 

However the issue implies that there may be some use case on type casting these boolean vector registers. We want to ask for more input on this before we decide to add more intrinsics API-s.

- What are the use cases where we want type conversions between different `vbool` values?
- What are the use cases of casting between boolean vector registers to LMUL<=1 values integer values?

nick-knight, 2022-11-30:
@jan-wassenberg [posted](https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/170#issuecomment-1332005024) a use-case of `vint`-to-`vbool` conversion: initializing a mask register to `0b0101...0101` (select even-indexed elements) via an X-to-V splat. This could be used for building other periodic sequences of mask bits (with power-of-two period).

jan-wassenberg, 2022-12-08:
Another use case: load-expand of vint8m8_t. We want to promote to u16 so we can use viota.m and then vrgatherei16, but LMUL is already maxed out. We can instead split both the input vector (easy) and mask (requires casting?) into upper and lower halves, run the LMUL<8 code, then concatenate.

nick-knight, 2022-12-08:
Thanks @jan-wassenberg. Just out of curiosity, can you provide any insight why trimming LMUL from the start (vint8m4_t) would be less satisfactory? I haven't done a careful analysis of the alternatives, but I don't see a clear-cut dynamic code size savings. (I'm not disputing the validity of your use case: this is indeed something that a programmer might want to do, and currently cannot.)

jan-wassenberg, 2022-12-08:
@nick-knight The Highway ops are defined/required to work for any vector LMUL, and our tests verify that. I agree m8 isn't the most sensible path here for someone who knows what is going on, but it's also non-obvious why 'vdecompress_u8m8' shouldn't be used. (Because viota_m might overflow 8-bit lanes, which isn't visible to users.)

jan-wassenberg, 2023-04-24:
Nice :)

PR #221 (2023-05-03): Define reinterpret intrinsics between vector boolean types and LMUL=1 (m1) vector integer types
Resolves #156.

The LLVM implementation is
D149642 [[RISCV] Support vreinterpret intrinsics between vector boolean type and m1 vector integer type](https://reviews.llvm.org/D149642)
closed 2023-05-03 by eopXD